olikraus/u8g2 #659: incompatibility when u8glib and u8g2 lib installed together?

i-make-robots, 2018-07-19:
I have a minimal u8g2 test working on a RUMBA board (mega2560 compatible).  When I add the relevant code to a larger project, I'm getting the errors below.  I've made sure that I'm not including LiquidCrystal or u8glib anywhere in the project.  Any idea what I'm doing wrong, please?

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_ClearPolygonXY'

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

c:/users/admin/documents/arduinodata/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_AddPolygonXY'

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_DrawPolygon'

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\Admin\Documents\Arduino\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

olikraus, 2018-07-19:
Somehow old u8glib and newer u8g2 are included. Do you need both? If not try to remove one of the libs.

i-make-robots, 2018-07-19:
I need one or the other for different projects.

i-make-robots, 2018-07-19:
Thank you for your time.  I switched to the older u8g library as it compiled error free.  Thank you!

closed 2018-07-19 by i-make-robots