olikraus/u8g2 #663: SSD1317 0.96 inch 96x96  7P SPI OLED Screen

szmolnar, 2018-07-25:
Hi!

I would like to use this screen for one of my projects, is it possible to add a constructor for it?
It is partially working with the SSD1306 constructor U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_MIRROR, /* clock=*/ 5, /* data=*/ 4, /* cs=*/ 15, /* dc=*/ 14, /* reset=*/ 13);
It is working mirrored and x is shifted +32 pixels, Y is shifted -32 pixels
Attaches picture was drawn with:
u8g2.drawFrame(32, -32, 96, 96);
u8g2.drawHLine(32, 1, 96);

[image]

Thanks!

olikraus, 2018-07-27:
ok, I have to look at this. Not sure when I will have time...

szmolnar, 2018-07-28:
Ok, many thanks in advance!

olikraus, 2018-08-23:
ok, I think this is implemented. e.g.: 
`U8G2_SSD1317_96X96_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`
Can you test this new constructor?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2018-08-24:
any feedback/comments?

olikraus, 2018-10-28:
no further feedback...

closed 2018-10-28 by olikraus

ako82, 2019-01-30:
Hi.
I had salvaged this display out of a broken e-cigarette and managed to get it running. But I also had the trouble that only the left side was addressable (64x96 to precise) no matter which offset i had used.
I don't know much about programming but after some research I figured out that the SSD1317 is in fact a 128x96 controller.
After some further research I found [this](http://www.iamoled.com/upload/file/20180119/20180119114600_9009.pdf) document (Figure 4.5) which helped me set the correct initial commands in the u8x8_d_ssd1317.c

Hoffe ich konnte ein kleines Bisschen was zu der Sache hier beitragen.
Ansonsten... Danke für all deine Mühen und deine tolle Arbeit!

olikraus, 2019-01-30:
wow, can you share the correct commands for u8x8_d_ssd1317.c? Or maybe paste/attach the modified file here.

Danke!

ako82, 2019-01-30:
> /*

  u8x8_d_ssd1317.c

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2018, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  SSD1317: 128x96 OLED

*/

#include "u8x8.h"

/* more or less generic setup of all these small OLEDs */
static const uint8_t u8x8_d_ssd1317_96x96_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  
  U8X8_C(0x0ae),		        /* display off */
  U8X8_CA(0x0d5, 0x0d1),		/* clock divide ratio (0x00=1) and oscillator frequency (0x8) */
  U8X8_CA(0x0a8, 0x05f),		/* multiplex ratio */
  U8X8_CA(0x0d3, 0x000),		/* display offset */
  U8X8_CA(0x0a2, 0x000),		/* set display start line to 0 */
  U8X8_CA(0x08d, 0x014),		/* [2] charge pump setting (p62): 0x014 enable, 0x010 disable, SSD1306 only, should be removed for SH1106 */
  U8X8_CA(0x020, 0x000),		/* page addressing mode */
  
  U8X8_C(0x0a0),			/* segment remap a0/a1*/
  U8X8_C(0x0c8),			/* c0: scan dir normal, c8: reverse */
  
  // Flipmode
  //U8X8_C(0x0a1),			/* segment remap a0/a1*/
  //U8X8_C(0x0c0),			/* c0: scan dir normal, c8: reverse */
  
  U8X8_CA(0x0da, 0x012),		/* com pin HW config, sequential com pin config (bit 4), disable left/right remap (bit 5) */

  U8X8_CA(0x081, 0x09f), 		/* [2] set contrast control */
  U8X8_CA(0x0d9, 0x0f1), 		/* [2] pre-charge period 0x022/f1*/
  U8X8_CA(0x0db, 0x0ff), 		/* vcomh deselect level */  
  
  // if vcomh is 0, then this will give the biggest range for contrast control issue #98
  // restored the old values for the noname constructor, because vcomh=0 will not work for all OLEDs, #116
  
  U8X8_C(0x02e),			/* Deactivate scroll */ 
[... 136 lines cut ...]
  /* flipmode_x_offset = */ 16,
  /* pixel_width = */ 96,
  /* pixel_height = */ 96,
};

uint8_t u8x8_d_ssd1317_96x96(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    
  if ( u8x8_d_ssd1317_generic(u8x8, msg, arg_int, arg_ptr) != 0 )
    return 1;
  
  switch(msg)
  {
    case U8X8_MSG_DISPLAY_INIT:
      u8x8_d_helper_display_init(u8x8);
      u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1317_96x96_init_seq);    
      break;
    case U8X8_MSG_DISPLAY_SETUP_MEMORY:
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ssd1317_96x96_display_info);
      break;
    default:
      return 0;
  }
  return 1;
}

ako82, 2019-01-30:
```
/*

  u8x8_d_ssd1317.c

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2018, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  SSD1317: 128x96 OLED

*/

#include "u8x8.h"

/* more or less generic setup of all these small OLEDs */
static const uint8_t u8x8_d_ssd1317_96x96_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  
  U8X8_C(0x0ae),		        /* display off */
  U8X8_CA(0x0d5, 0x0d1),		/* clock divide ratio (0x00=1) and oscillator frequency (0x8) */
  U8X8_CA(0x0a8, 0x05f),		/* multiplex ratio */
  U8X8_CA(0x0d3, 0x000),		/* display offset */
  U8X8_CA(0x0a2, 0x000),		/* set display start line to 0 */
  U8X8_CA(0x08d, 0x014),		/* [2] charge pump setting (p62): 0x014 enable, 0x010 disable, SSD1306 only, should be removed for SH1106 */
  U8X8_CA(0x020, 0x000),		/* page addressing mode */
  
  U8X8_C(0x0a0),			/* segment remap a0/a1*/
  U8X8_C(0x0c8),			/* c0: scan dir normal, c8: reverse */
  
  // Flipmode
  //U8X8_C(0x0a1),			/* segment remap a0/a1*/
  //U8X8_C(0x0c0),			/* c0: scan dir normal, c8: reverse */
  
  U8X8_CA(0x0da, 0x012),		/* com pin HW config, sequential com pin config (bit 4), disable left/right remap (bit 5) */

  U8X8_CA(0x081, 0x09f), 		/* [2] set contrast control */
  U8X8_CA(0x0d9, 0x0f1), 		/* [2] pre-charge period 0x022/f1*/
  U8X8_CA(0x0db, 0x0ff), 		/* vcomh deselect level */  
  
  // if vcomh is 0, then this will give the biggest range for contrast control issue #98
  // restored the old values for the noname constructor, because vcomh=0 will not work for all OLEDs, #116
  
[... 141 lines cut ...]
};

uint8_t u8x8_d_ssd1317_96x96(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    
  if ( u8x8_d_ssd1317_generic(u8x8, msg, arg_int, arg_ptr) != 0 )
    return 1;
  
  switch(msg)
  {
    case U8X8_MSG_DISPLAY_INIT:
      u8x8_d_helper_display_init(u8x8);
      u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1317_96x96_init_seq);    
      break;
    case U8X8_MSG_DISPLAY_SETUP_MEMORY:
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ssd1317_96x96_display_info);
      break;
    default:
      return 0;
  }
  return 1;
}
```

I have used your file and modified the basic initial setup part according to the PDF and I had to modify the default- and flipmode offsets with +16 each.

olikraus, 2019-01-30:
Nice, thanks. Did you also test the FlipMode.ino page mode example? Will the border frame always at the same place?

ako82, 2019-01-30:
Hmm...
Can be that I copied it out of the Flipmode.ino, I am not sure, but I ended up with this code for testing and at the end, the frame was always at the same place

```
void loop(void) {

  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFlipMode(1);
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawFrame(0, 0, 96, 96);
  u8g2.drawStr(3,11,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1500);

  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFlipMode(1);
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawFrame(0, 0, 96, 96);
  u8g2.drawStr(3,11,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1500);  

}
```

olikraus, 2019-01-30:
thanks, added to the repo.