olikraus/u8g2 #665: drawLine wrong pixel count on x

StoneAgeSkillz, 2018-07-28:
Hi,
i'm using sdd1306 based 128*32 i2c OLED with ESP12e. If i use drawFrame(0, 31, 127, 31) it draws it a pixel short. If the count starts at 0 and is 128px long, shouldn't it end at 127, not 128?
It's not a big issue, more a source of confusion.

Have a nice day.

olikraus, 2018-07-31:
you have a picture?
Which constructor did you use?

olikraus, 2018-07-31:
ok, thanks. I will do some tests. Hope I can reproduce this.

olikraus, 2018-08-22:
Any datasheet for the display avaialble? It looks more like a 128x36 OLED (instead of a 128x32 pixel version).
How does normal text look like? Can you post a version of the HelloWorld.ino output?

StoneAgeSkillz, 2018-08-24:
Hi,
sorry for the delay. I don't have a datasheet, maybe the seller can provide one. 
This is the OLED i bougth:
https://www.ebay.com/itm/0-91-IIC-I2C-128x32-White-OLED-LCD-Display-Module-For-Arduino-PIC/172435137027?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

And here is the requested output:
[image]

Don't mind the pcb, it's the same schematic, but with ESP soldered on the other side.

olikraus, 2018-08-24:
Hmmm the last picture looks good. Does the problem still appear?

StoneAgeSkillz, 2018-08-24:
It does. Same code as before:
[image]

StoneAgeSkillz, 2018-08-24:
Ok, i officialy admit i am an idiot and have wasted your time. I used another lib before yours and there was an X-start, Y-start, X-end, Y-end, to mark the corners. I did read your wiki, but too fast... Now i read it again and found the issue. It was me. I wasted your time. I am very, very sorry for this. Dishonor on me, my family, my cow...

closed 2018-08-24 by StoneAgeSkillz

olikraus, 2018-08-24:
hehe, no issue. I came to the same result as you (8 minutes later).
Indeed u8g2 uses width and hight arguments:

https://github.com/olikraus/u8g2/wiki/u8g2reference#drawframe