olikraus/u8g2 #666: link to download the latest Arduino u8g2 library

mikekgr, 2018-07-30:
Dear Sirs,
can you provide me the link to download the latest Arduino u8g2 library? Can be this link permanent to point always to the latest "snapshot" of the u8g2 Arduino library?

Many thanks and Best Regards,
Mike Kranidis

v1993, 2018-07-30:
Is https://github.com/olikraus/u8g2/archive/master.zip ok for you?

mikekgr, 2018-07-30:
Dear @v1993  yes this is what I am looking for. Thanks!

olikraus, 2018-07-31:
Actually, the Arduino library is available here for download:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

mikekgr, 2018-07-31:
... Small detail but now I realized that this issue is #666  ...   to whom may concern !

olikraus, 2018-07-31:
hehe... ok, i am more interessted in number theorie... the sum of the first (6+6)*(6+6) digits of Pi is 666.

closed 2018-08-04 by olikraus