olikraus/u8g2 #669: Font with same pixel width for every char

sergueiesp, 2018-08-02:
Hello, I'm using a 1.5" OLED and i would like to use a medium font (around 30px) that has the same pixel width for every char. 

I mean, i am writting a float numer with 2 decimal precision, but the decimal separator is not in the same pixel. It moves right or left depending on the other numbers. 

Font u8g2_font_px437wyse700b_mn works fine, but is too little for my purpose.

Thank you very much.

olikraus, 2018-08-02:
In general you could select any font where the last to last char is "m".
If such a font is not there, then it can be created.
What about this font:
[image]

sergueiesp, 2018-08-02:
Thank you for your fast response. I have attaches a Zip file with 2 videos. In one you can see how it is working the font you have proposed. In the other, using u8g2_font_px437wyse700b_mn , how i would like to perform.

Thank you very much,
[Video.zip](https://github.com/olikraus/u8g2/files/2252692/Video.zip)

olikraus, 2018-08-03:
ok, I see the problem...

olikraus, 2018-08-03:
hmmm.... the glyph width seems to be different:
u8g2_GetGlyphWidth(&u8g2, '4') --> 19
u8g2_GetGlyphWidth(&u8g2, '9') --> 20

olikraus, 2018-08-04:
I have created beta 2.23.13 which should fix this problem. Can you test this version?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

sergueiesp, 2018-08-09:
First of all, I'm sorry i couln't test the lib until today.

Thank you very much, tested with _u8g2_font_inb30_mn_ font and works perfect!

Again, Thank You very much.

olikraus, 2018-08-11:
Thanks for your feedback!

olikraus, 2018-08-23:
done (also documented in ChangeLog)

closed 2018-08-23 by olikraus