olikraus/u8g2 #670: Hardware SPI not working for EA DOG XL 240

n3krO, 2018-08-02:
Hello, I'm trying to get EA DOG XL 240 to work with Hardware SPI and it seems like it's not working.

When using software spi I use the same pins as if it was hardware spi:

U8G2_UC1611_EA_DOGXL240_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 52, /* data=*/ 51, /* cs=*/ 43, /* dc=*/ 49, /* reset=*/ 48);   // Due, SW SPI, DOGXL240 Test Board

So then I try to use hardware spi with the same pins (so I dont have to change anything physically) and the screen stops working....

U8G2_UC1611_EA_DOGXL240_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 43, /* dc=*/ 49, /* reset=*/  48);

I tried to define SS as OUTPUT to make sure the Hardware SPI chip wasn't getting disabled by a random value at SS.

Note: I'm using Arduino Mega.

olikraus, 2018-08-03:
Do you use any level shifter between Arduino Board and the DOG XL 240?

n3krO, 2018-08-03:
No I don't.

I have the DOG XL 240 VDD's connected to Arduino's 3.3 V and VSS to GND.

But the SPI pins are all connected directly from the digital pins to the DOG XL 240 pins.

Do I need a level shifter when working with hardware SPI but not when using software SPI?

olikraus, 2018-08-03:
> I have the DOG XL 240 VDD's connected to Arduino's 3.3 V and VSS to GND.

Sure, but the GPIO outputs of the Mega still have a high level of 5V.

> Do I need a level shifter when working with hardware SPI but not when using software SPI?

Most probably yes. I suggest to use a 74HC4050.
Actually you would need a level shifter in both cases, but for hardware SPI it is much more critical.

n3krO, 2018-08-03:
Thank you sir. Can you confirm me that the constructor for HW SPI is correct? I couldn't find it in the example code so I had to search through the library for it.

olikraus, 2018-08-03:
I do not know your exact wiring, but at least I did not find a mistake in the constructor.

n3krO, 2018-08-07:
So I just got some level shifters and it's not working anyway.

Same as before, when using SW_SPI it works and when using HW_SPI it doesn't do anything.

There are my constructors:

[code]U8G2_UC1611_EA_DOGXL240_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 52, /* data=*/ 51, /* cs=*/ 53, /* dc=*/ 49, /* reset=*/ 48);   // Due, SW SPI, DOGXL240 Test Board

U8G2_UC1611_EA_DOGXL240_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 53, /* dc=*/ 49, /* reset=*/  48);[/code]

Right now I shouldn't have to change the wiring when going from SW to HW....

olikraus, 2018-08-07:
At this point, I personally would start to investigate signals more closely with my scope.