olikraus/u8g2 #671: how to use open_iconic

sarwarislammoon, 2018-08-03:
Hi
I am new with GLCDs can u please tell me how can i print custom symbols like wifi in my lcd . I am using ST7565 lcd , i can write text on it using U8g2 , but i am very much confused with .

after a lot of trial and error with U8g2font.c I found that following code print wifi symbol , but is there any simple way to print it ? or at least how can I find strings corrospond with symbol .

thanks in advance .

code :
u8g2.setFont(u8g2_font_open_iconic_www_1x_t);
u8g2.setCursor(90,63);
u8g2.print("Q\23\210\24");

olikraus, 2018-08-03:
Let's take the 2x as reference (because of readability here):
[image]

I assume you mean that icon at the 9th position in the first row, correct?
Dezimal: Take the left most value (which is 64) and and the position (which is 9). You will get 73.
Hexadecimal. Take the left hex number (0x040) add the position (again 9), you get 0x049.

Then use the drawGlyph function:
```
u8g2.setFont(u8g2_font_open_iconic_www_1x_t);
u8g2.setCursor(90,63);
u8g2.drawGlyph(74);
```
or
```
u8g2.setFont(u8g2_font_open_iconic_www_1x_t);
u8g2.setCursor(90,63);
u8g2.drawGlyph(0x49);
```

olikraus, 2018-08-03:
Oh, I think you mean the second icon in the second row. Then its 81, right?
```
u8g2.setFont(u8g2_font_open_iconic_www_1x_t);
u8g2.setCursor(90,63);
u8g2.drawGlyph(81);
```

sarwarislammoon, 2018-08-03:
Thanks a lot .............. just one little query  , is "u8g2.drawGlyph()"; needs co-ordinate as its argument , i mean u8g2.drawGlyph(90,63,81);

olikraus, 2018-08-03:
ah yes, of course, you need to pass the position first. 

https://github.com/olikraus/u8g2/wiki/u8g2reference#drawglyph