olikraus/u8g2 #673: SSD1322 upside down

brunoeagle, 2018-08-06:
Hi,

The display is this one and the controller is SSD1322: https://www.buydisplay.com/default/serial-oled-module-price-3-2-inch-display-256x64-screens-white-on-black

Don't know if I'm doing something wrong, but the image is upside down:

[image]

The display is using 8080 parallel interface, and the code for showing this image is:

`		u8g2_Setup_ssd1322_nhd_256x64_f( &display, U8G2_R0, cb1, cb2);
		u8g2_InitDisplay( &display ); // send init sequence to the display, display is in sleep mode after this,
		u8g2_SetPowerSave( &display, 0 ); // wake up display
		u8g2_ClearDisplay( &display );
		u8g2_SetFont( &display, u8g2_font_roentgen_nbp_tf );
		u8g2_DrawBox( &display, 0, 0, 10, 5 );
		u8g2_DrawBox( &display, 0, 6, 10, 5 );
		u8g2_SendBuffer( &display );`

cb1:
`uint8_t cb1(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	uint8_t *data;

	switch(msg){
		case U8X8_MSG_BYTE_SEND:
			data = (uint8_t *)arg_ptr;
			while( arg_int > 0 ) {
				GPIOK->ODR = *data;
				//u8x8_gpio_call( u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->data_setup_time_ns );
				asm("NOP");
				WR_LOW;
				//u8x8_gpio_call( u8x8, U8X8_MSG_GPIO_E, 0 );
				asm("NOP");
				//u8x8_gpio_call( u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->write_pulse_width_ns );
				WR_HIGH;
				//u8x8_gpio_call( u8x8, U8X8_MSG_GPIO_E, 1 );
				data++;
				arg_int--;
			}
			break;

		case U8X8_MSG_BYTE_INIT:
			//u8x8_gpio_SetCS( u8x8, u8x8->display_info->chip_disable_level );
			CS_HIGH;
			WR_HIGH;
			//u8x8_gpio_call( u8x8, U8X8_MSG_GPIO_E, 1 );
			break;

		case U8X8_MSG_BYTE_SET_DC:
			u8x8_gpio_SetDC( u8x8, arg_int );
			break;

		case U8X8_MSG_BYTE_START_TRANSFER:
			u8x8_gpio_SetCS( u8x8, u8x8->display_info->chip_enable_level );
			u8x8->gpio_and_delay_cb( u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
			break;

		case U8X8_MSG_BYTE_END_TRANSFER:
			u8x8->gpio_and_delay_cb( u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL );
			u8x8_gpio_SetCS( u8x8, u8x8->display_info->chip_disable_level );
			break;

		default:
			return 0;
	}
	return 1;
}`

cb2:
`uint8_t cb2(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	GPIO_InitTypeDef GPIO_InitStruct;
	uint8_t i;
	switch(msg) {
		case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8

			__HAL_RCC_GPIOG_CLK_ENABLE();
			__HAL_RCC_GPIOK_CLK_ENABLE();

			// DB7~DB0 lines
			GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
			GPIO_InitStruct.Pull = GPIO_PULLUP;
			GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
			GPIO_InitStruct.Pin = GPIO_PIN_All;
			HAL_GPIO_Init( GPIOK, &GPIO_InitStruct );

			// RD, WR, DC, RST, CS respectively
			GPIO_InitStruct.Pin = GPIO_PIN_9 | GPIO_PIN_10 |
						GPIO_PIN_11 | GPIO_PIN_12 | GPIO_PIN_13;
			HAL_GPIO_Init( GPIOG, &GPIO_InitStruct );
			RD_HIGH;
			break;

		case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
			break;

		case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
			for(i = 0; i < 25; i++ ) {
				asm("NOP");
			}
			break;

		case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
			for(i = 0; i < 200; i++ ) {
				asm("NOP");
			}
			break;

		case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
			vTaskDelay( 1 / portTICK_PERIOD_MS );
			break;
    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
    	if( arg_int )
    		WR_HIGH;
    	else
    		WR_LOW;
    	break;

    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
    	if( arg_int )
    		CS_HIGH;
    	else
    		CS_LOW;
    	break;

	case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
    	if( arg_int )
    		DC_HIGH;
    	else
    		DC_LOW;
		break;

	case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
    	if( arg_int )
    		RST_HIGH;
    	else
    		RST_LOW;
		break;

	case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
    	if( arg_int )
    		CS_HIGH;
    	else
    		CS_LOW;
		break;

    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;
}`

When I initialize it with U8G2_R2, the rotation fix, but I have an offset in x of ~10 pixels or so...

Thanks in advance!

olikraus, 2018-08-06:
Hmmm... did you try https://github.com/olikraus/u8g2/wiki/u8g2reference#setflipmode ?

The 10 pixel offset sounds strange.

brunoeagle, 2018-08-06:
The SetFlipMode indeed works! thanks!

But I was noticing something wierd, I'm drawing 2 boxes now:
`u8g2_DrawBox( &display, 0, 0, 255, 5 );
u8g2_DrawBox( &display, 0, 6, 10, 5 );`

without the FlipMode enabled, the image is like this:
[image]

with the FliMode enabled, like this:
[image]

Look how the end of the bigger(255) box never touches the display border like in the beggining, both with flip mode and not..

Did you noticed in the pictures this?

Regards!

brunoeagle, 2018-08-06:
To be precise, the last 8 pixels of the box are not being drawn...

olikraus, 2018-08-06:
Did you enable 16 bit mode? 
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

brunoeagle, 2018-08-06:
Is working properly now with 16bit mode. Didn't know that I had to enable... 
Maybe a suggestion to add it to https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform ...?

Thanks man!!

olikraus, 2018-08-06:
good point...

olikraus, 2018-08-23:
done

closed 2018-08-23 by olikraus