olikraus/u8g2 #676: Problem building big logisoso fonts with € symbol

MMartinezV, 2018-08-09:
Hi,

Following your instructions on closed issue #608  I've managed to generate many sizes of logisoso to use the € symbol on an e-ink 2.13" display using U8g2_for_Adafruit_GFX.

After generating it I simply add it to u8g2_fonts.c and the header to U8g2_for_Adafruit_GFX.h.

I've found two problems doing it:

1) If I use the command to generate the logisoso20+€ as described here (using -m "32-127,8364") it doesn't work for me. Otherwise, if I use (-m "32-8364") then I'm able to use the € symbol. It is all right as it works but this way the c array is bigger so I would prefer to select less symbols and keep the array smaller.

The commands I've used is:

`otf2bdf/otf2bdf -a -r 72 -p 30 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '32-8364' tmp.bdf -o u8g2_font_logisoso20_tx.c  -n u8g2_font_logisoso20_tx -d ./bdf/helvB18.bdf`

2) I've been able to generate logisoso 20pt, 24 and 28 and they work fine, but I have problems with 30, 32, 34 and 42pt. After generating them, when I try to use these fonts on an ESP32 device it hangs. I've tought that the problem could be the size of the array but I'm able to use a standard (without € symbol) u8g2_font_logisoso50_tf (9491 elements in the font array) that has more elements than the 34pt font.

For more information these are the commands that I'm using to build the fonts:

`otf2bdf/otf2bdf -a -r 72 -p 43 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '30-8364' tmp.bdf -o u8g2_font_logisoso30_tx.c  -n u8g2_font_logisoso30_tx -d ./bdf/helvB18.bdf`

`otf2bdf/otf2bdf -a -r 72 -p 45 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '32-8364' tmp.bdf -o u8g2_font_logisoso32_tx.c  -n u8g2_font_logisoso32_tx -d ./bdf/helvB18.bdf`

`otf2bdf/otf2bdf -a -r 72 -p 49 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '8364,32-127' tmp.bdf -o u8g2_font_logisoso34_tx.c  -n u8g2_font_logisoso34_tx -d ./bdf/helvB18.bdf`

I've also tested using ./bdf/helvB24.bdf with same results.

Can you help me to build at least the 34pt font?

Thanks,

  Manuel Martínez

olikraus, 2018-08-11:
Just to ensure that there are no other problems, you must use the old bdfconv version for U8g2 for Adafruit:
https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdfconv_2_22.exe

Also -m "32-127,8364" shell work. You could also try to put this option into a file and use -M

olikraus, 2018-08-21:
Any updates? Can I close this issue?

MMartinezV, 2018-08-22:
Sorry, I've been on holidays last week. Tomorrow I'll check it again.

MMartinezV, 2018-08-28:
Yes, I was using the old bdfconv version compiled on Ubuntu 18.04.1 LTS.

Same results with -m '32-127,8364' and -M symbol_map.txt with symbol_map.txt contents: 32-127,8364

I'll do some more tests.

Regards,

  Manuel

olikraus, 2018-10-28:
Note: I updated U8g2_for_Adafruit_GFX. The font format should be identical now.

MMartinezV, 2018-11-07:
Now it works.

Thanks!

olikraus, 2018-11-07:
The compressed bitmap data of one glyph must not exceed 255 bytes. In your case the "@" glyph seesm to be too large. You can remove this from the conversion process if you do not need "@":
 
`-m '32-63,65-127,8364'`

MMartinezV, 2018-11-07:
It works, thanks again.

Regards,

  Manuel

closed 2018-12-20 by olikraus