olikraus/u8g2 #677: AMOLED CONTROLER RM67162

ELIVB, 2018-08-10:
Hello
I am in the advanced stages of building a project on the basis of Adafruit feather nRF52832, The project will be minimized to size of a watch. In recent days I'm starting to check what type of screen to use.

https://www.alibaba.com/product-detail/1-39-400x400-full-color-circular_60817940872.html?spm=a2700.icbuShop.prewdfa4cf.14.647461a6bu9xDi

I contacted the manufacturer and he sent me several files (Data Sheet) but without librarys to communicate with the screen.
 
Do you think you will publish a library for AMOLED?

Thanks
eli

olikraus, 2018-08-10:
This seems to be a RGB display. U8g2 only supports monochrome displays.

ELIVB, 2018-08-11:
Hello
After a thorough acquaintance with the screens, I want to hear your opinion, do you think that arduino 64Mz like Adafruit feather nRF52832 Soc can handle the AMOLED In terms of data traffic and speed?

Can you build complete library for AMOLED? Of course for payment

eli

olikraus, 2018-08-11:
>  do you think that arduino 64Mz like Adafruit feather nRF52832 Soc can handle the AMOLED In terms of data traffic and speed?

Traffic: yes, speed: no

You AMOLED requires 400*400*3/8 = 60000 bytes. To have suitable satisfying speed you need the complete display buffer in the RAM area of the Adafruit feather. Ok, you have 64K available, but then the complete RAM will be used for the display RAM buffer. If the remaining 4K is sufficient for your application, then speed will be ok, otherwise not.

ELIVB, 2018-08-11:
Hello
The Adafruit nrf52 as **512KB flash** and **64KB SRAM** and **running at 64MHz**

I want to point out that I have no knowledge of electronics, Do you think there is a third way that it should work?

eli

ELIVB, 2018-08-11:
[image]

olikraus, 2018-08-11:
> Do you think there is a third way that it should work?

Depends on what you or your users expect under "should work". If flickering and low speed is acceptable, then you can write directly to the display.

vin7102, 2018-08-11:
@ELIVB 

If your not entirely limited to the nRF52, you can look at the Adafruit M4 series which boasts 512 KB flash, 192 KB RAM and runs at 120MHz.  (similar form factor to the nRF52)
Being a watch type project though, you may be looking for integrated bluetooth, of which you may have to wait for Adafruit to release an M4 series "Bluefruit" or somehow fit a bluetooth breakout into your design.

Regards...

olikraus, 2018-08-13:
Update: I made a mistake in the memory calculation. Actually, about 470KB are required.