olikraus/u8g2 #679: AMOLED 1.39 issue #677

ELIVB, 2018-08-13:
Hello
I hope it's all right to contact you again, I wrote to you few days ago, issue #677. I wrote to the company that manufactures the screens:
My question was:
I did a lot of inquiries yesterday, I would like you to check with your engineers.
I use a microcontroller for my project the adafruit nrf52832 Soc.  
ARM Cortex M4F (with HW floating point acceleration) running at 64MHz 512KB flash and 64KB SRAM.
We noted in forums that the controller is too weak to your AMOLED 400X400 So I'd like to ask what your engineers think. what the processor speed should be and how much memory.
The answer is:
Since we are mainly specialized in TFT and not professional in microcontrollers, we cannot provide professional advice.
In terms of rate, if 1.39-inch use CMD mode, the transfer chip of SSD2805 is used on our side, and the MCU interface is transferred to MIPI interface. If the rate is calculated according to McU-8bit, 400x400x3x60(Hz) = 28.8mhz. CMD Mode, Driver IC of 1.39 is shipped with GRAM, as follows
400RGB X 400 with 400x24bit X 400 GRAM

New product from adafruit: 
"This feather is powered by our new favorite chip, the ATSAMD51J19 -  with its 120MHz Cortex M4 with floating point support and 512KB Flash and 192KB RAM."

Do you think that with the new data the screen should work very well?

I believe that technological evolution cannot be stopped, there will always be faster processors and bigger memories for microcontrollers.

I would like to thank you in advance
eli

olikraus, 2018-08-13:
https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51

Yes, I think this has at least the required RAM on board...

ELIVB, 2018-08-13:
Hi,

All right, I hear good news from you.

Thanks
eli

olikraus, 2018-08-13:
I guess, it will be complicated: https://blog.adafruit.com/2016/03/31/control-smartphone-display-from-mipi-display-serial-interface-arduino-and-raspberrypi-compatible/

So as a result, I have doubts again, that the feather m4 express is powerful enough. Didn't you analyse whether parts fit together?

olikraus, 2018-08-13:
I think I was also wrong with my memory estimation. Actually you need around 500KB for the display.
The feather m4 express is by far not powerful enough.

ELIVB, 2018-08-13:
Hello 
Thanks for coming back.
The Truth There are many things that I do not know but need to check out, I'm looking for answers and I really appreciate the help.

SPI to MIPI D-PHY

http://www.latticesemi.com/Products/DesignSoftwareAndIP/IntellectualProperty/ReferenceDesigns/ReferenceDesign04/SPItoMIPIDPHY

This is a bridge, Eric from Nordic Said it could work. I find Other chip from Toshiba.

https://www.mouser.co.il/new/Toshiba-Semiconductors/toshiba-tc358778xbg-dsi/

About the memory, How did you arrive to 500k?

Thanks

eli

olikraus, 2018-08-13:
Memory: 400 pixel x 400 pixel x 3 bytes per pixel = 480 000 Bytes = 468 KB

Any use of SPI during datatransfer between controller and display will slow down the refresh a lot.

ELIVB, 2018-08-14:
Hello
I really hope you understand me, my English is not the best.
I've already mentioned that I'm not an electrician, but I work with common sense and may have a solution.
You are a programmer who knows how to program hardware, what do you say about the next idea, is it possible?
I switched to a different processor for my project the ATSAMD51J19 - with its 120MHz Cortex M4 but I also use the Nordic nRF52832 or nRF52840 this chip is Soc (System on chip - 32-bit ARM® Cortex®-M4F 64 MHz) this chip is integrated with Bluetooth chip.
I use only the Bluetooth so the Soc is free to use
In datasheet of the Nordic Nrf52840 in page 661 –
"The QSPI peripheral provides support for communicating with an external flash memory device using SPI"
Maybe I can add fast flash memory to Soc and the screen (AMOLED) work Against the Nordic chip.
In computer systems, the video card also has a processor and memory, and there is no burden to the main processor.
Do you think this is feasible?
Thanks
Eli

ELIVB, 2018-08-14:
Of course, everything is theoretical and if you think it can work, so I'll try at most I'll lose $ 300. Not a big deal

olikraus, 2018-08-14:
I am only a hobby programmer. And you claim you are not a specialist on this domain. 

I suggest not to do this project. Or at least choose a smaller display with 160x160 pixel max.

ELIVB, 2018-08-14:
Okay buddy, thank you very much for your time.
I'm one of those people who do not give up.

God bless you

eli

closed 2018-08-14 by ELIVB

olikraus, 2018-08-14:
Good luck