olikraus/u8g2 #680: Suddenly blank display

mkovero, 2018-08-15:
Hi, first of all thanks for your excellent work, I thought asking here if anyone happens to know where to look, basic chinese SSD1306 OLED previously worked fine with arduino is having problems with teensy 3.5 now.
Display first initializes fine but roughly after 15sec it turns blank / turns off.

This happens with example sketch too, for example:

```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 9, 5);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,15,"Hello World!");
  u8g2.sendBuffer();
  delay(1000);
}
```

Also VCOMH0/ALT0 variations work and behave similarly.
Anyone know whether this might be signaling issue (5v on arduino worked fine and 3.3v on teensy not being enough for CS/RST?), if so, would it be possible to workaround preferably without using logic level changers.
I have analog discovery 2 on my disposal so I can dig deeper I just would wish to know where to start looking.

Tyvm!:)

olikraus, 2018-08-15:
Hmmm .. Sounds like a faulty display.

olikraus, 2018-08-21:
Can we close this issue?

mkovero, 2018-08-21:
Yes, I'll leave a note whether I'm able to cope this with lowering VCC. thank you for your help!:)

closed 2018-08-21 by mkovero

mkovero, 2018-08-25:
for future reference, this was signal issue, RES wire was having bad time.