olikraus/u8g2 #682: Hardware I2C not working with OLED 128x128, SH1107

thucao, 2018-08-16:
Hi, 
I am working with a SEEED OLED Display 1.12" V2.2 128x128 with the SH1107 driver. I followed [#501](https://github.com/olikraus/u8g2/issues/501) and was able to get my OLED display working with the constructor
` U8G2_SH1107_128X128_1_SW_I2C u8g2 (U8G2_R0, SCL, SDA, U8X8_PIN_NONE); `
and changing the offset in the sh1107.c file to 
`
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
`
However, the display showed a black screen when I tried to use the constructor 
` U8G2_SH1107_128X128_1_HW_I2C u8g2 (U8G2_R0, U8X8_PIN_NONE); `

Yet, when I attempted to use the constructor ` U8G2_SSD1327_SEEED_96X96_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); ` the display did show something, even though it wasn't showing the right thing. 

I am using an Arduino Uno board.
Here is the datasheet for my OLED Display: 
[OLED Module;1.12 inch-White-27_38.9_1.28mm_Datasheet.pdf](https://github.com/olikraus/u8g2/files/2294564/OLED.Module.1.12.inch-White-27_38.9_1.28mm_Datasheet.pdf)

Thank you.

olikraus, 2018-08-16:
Strange. I have no proper access to the repo. Please let me know if you have any new ideas.

olikraus, 2018-08-21:
I have compared the setup sequence from the datasheet with the init sequence used by `U8G2_SH1107_128X128_1_SW_I2C`. It seems to be identical, so it should work.
Can you send a picture with the original offset?

veeppop, 2018-09-05:
HI,
  I also have the same question about using SEEED OLED Display 1.12" V2.2 128x128 with the SH1107 driver. And I download the latest library-"u8g2". 
When I change the code
`U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);` to `U8G2_SH1107_128X128_1_SW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);`,

it shows like this 
> "no matching function for call to 'U8G2_SH1107_128X128_1_SW_I2C::U8G2_SH1107_128X128_1_SW_I2C(const u8g2_cb_t*, int)'"

However , It never show something when i download the code to my board, BTW, I also use the Arduino Uno.

AnHardt, 2018-09-05:
@veeppop 
> `U8G2_SH1107_128X128_1_SW_I2C u8g2 (U8G2_R0, SCL, SDA, U8X8_PIN_NONE);`
vs your
> ` U8G2_SH1107_128X128_1_SW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);`

The difference is two more pin arguments.
At a Uno (and most other boards) there is only one set of **hardware** I2C pins. For the **software** I2C you have to define them.

veeppop, 2018-09-05:
@AnHardt 
Hey, thanks for your notice. I've tried to add the SCL, SDA pin, and use the code like
`U8G2_SH1107_128X128_1_SW_I2C u8g2 (U8G2_R0, /* SCL=*/ 6, /* SDA=*/5, U8X8_PIN_NONE);`
It's still not working.

olikraus, 2018-09-05:
Wiring  problem? Wrong display? Can you link the datasheet of your display?

veeppop, 2018-09-06:
This is all the files that I can found from https://www.seeedstudio.com/Grove-OLED-Display-1-12-V2-p-3031.html
[OLED.Module.1.12.inch-White-27_38.9_1.28mm_Datasheet.pdf](https://github.com/olikraus/u8g2/files/2355252/OLED.Module.1.12.inch-White-27_38.9_1.28mm_Datasheet.pdf)
[Grove - OLED Display 1.12'' V2_Eagle File.zip](https://github.com/olikraus/u8g2/files/2355253/Grove.-.OLED.Display.1.12.V2_Eagle.File.zip)

olikraus, 2018-09-06:
Quote from initial post:

> Yet, when I attempted to use the constructor
`U8G2_SSD1327_SEEED_96X96_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`
the display did show something, even though it wasn't showing the right thing.

Can you also see something with this constructor?

Will there be anything visible with `U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);`?

veeppop, 2018-09-06:
I've tried it, but it doesn't show anything.

closed 2018-10-28 by olikraus