olikraus/u8g2 #683: STM32 / New Heaven NHD-3.12-25664UCB2

meldano, 2018-08-19:
Hi,
I will control a NHD-3.12-25664UCB2 (Chip SSD1322) Display with the great u8g2 lib.
The Display support 3- / and 4- wire SPI.

In case of 3-wire SPI it is:
-/Reset
-/CS
-SDIN
-SCLK

In case of 4-wire SPI
-/Reset
-/CS
-DC
-SDIN
-SCLK

I read the "Porting to new MCU platform" and all other articles I found

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

The following is not clear for my understanding:

a) Is it possible to run the display in both modes (3-wire/4-wire SPI) with the u8g2 lib? 
Where can I "tell" the mode to the lib?

b) Wich Messages are necessary in the "GPIO and Delay callback" when I use the STM32 Hardware SPI?

`uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
...
}
`

I have implemented the following function for Hardware SPI

`uint8_t u8x8_byte_stm32_hal_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t byte;
	uint8_t *data;
 
	switch (msg)
	{
	case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while (arg_int > 0)
		{
			byte = *data;
			data++;
			arg_int--;
			HAL_SPI_Transmit(&hspi1, &byte, 1, 100);
		}
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_SET);
		__asm volatile("nop");
		
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:
		
		__asm volatile("nop");
		HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_RESET);
		break;
	default:
		return 0;
	}
	return 1;
}`

And in my opinion the GPIO an Delay callback

`uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	switch (msg)
	{
		
		case U8X8_MSG_GPIO_DC:			// Reset pin: Output level in arg_int
			if(arg_int)
				HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_SET);
			else
				HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_RESET);
		break;
		
		case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
			if(arg_int)
				HAL_GPIO_WritePin(SPI1_RESET_GPIO_Port, SPI1_RESET_Pin, GPIO_PIN_SET);
			else
				HAL_GPIO_WritePin(SPI1_RESET_GPIO_Port, SPI1_RESET_Pin, GPIO_PIN_RESET);
		break;
	}

	return 1; // command processed successfully.
}`

Init the Display
`u8g2_Setup_ssd1322_nhd_256x64_1(&u8g2, U8G2_R0, u8x8_byte_stm32_hal_hw_spi, u8g2_gpio_and_delay_stm32);`

I hope somebody can help me to use the u8g2 and understand the usage!

Thanks
Daniel

olikraus, 2018-08-20:
You must also implement the delay messages. 

For 3w spi you need to transmit 9 bit instead of 8. It is a little bit more complicated, so I suggest to implement 4w spi first.

meldano, 2018-08-20:
Hi Mr. Kraus,

thank you for your answer!
I changed the code and will use 4-wire SPI.

Tommorow I will solder a test board.

`uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	switch (msg)
	{
		
	case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
	// can be used to setup pins
	break;		
		
	case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
		__asm volatile("nop");	
		break;    
		
	case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
		delay_ns(arg_int * 100);
		break;
		
	case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
		delay_ns(arg_int * 10000);
		break;
		
	case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
		HAL_Delay(arg_int);
		break;

		
		case U8X8_MSG_GPIO_DC:			// Reset pin: Output level in arg_int
			if(arg_int)
				HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_SET);
			else
				HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_RESET);
		break;
		
		case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
			if(arg_int)
				HAL_GPIO_WritePin(SPI1_RESET_GPIO_Port, SPI1_RESET_Pin, GPIO_PIN_SET);
			else
				HAL_GPIO_WritePin(SPI1_RESET_GPIO_Port, SPI1_RESET_Pin, GPIO_PIN_RESET);
		break;
	}

	return 1; // command processed successfully.
}`

meldano, 2018-08-22:
Hi,

my implementation doesn´t work.
A look in the datasheet of the Display said RESET and CS is inverted.

Is it necessary to Invert the logic in the funtions?

For example RESET

`case U8X8_MSG_GPIO_DC:			// Reset pin: Output level in arg_int
		if(arg_int)
			HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_RESET);
		else
			HAL_GPIO_WritePin(SPI1_DC_GPIO_Port, SPI1_DC_Pin, GPIO_PIN_SET);
	break;
`

olikraus, 2018-08-22:
No, for your display U8g2 already assumes low active signals. You sould in fact output exactly the level, which is requested in `arg_int`
In the above procedure, you have implemented DC pin, but i do not see CS pin... hope it is also there.

meldano, 2018-08-22:
I have to implement CS Pin in

`uint8_t u8x8_byte_stm32_hal_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)`

 procedure?

meldano, 2018-08-22:
I copied the procedure from other u8g2 stm32 issues.
I all examples i can´t find CS Pin Handling. Only in the 

`gpio_and_delay`

procedure.

meldano, 2018-08-23:
Here is the link

https://github.com/olikraus/u8g2/issues/356

rokk0 commented on Nov 6 2017

olikraus, 2018-08-24:
But there the `U8X8_MSG_GPIO_CS1` line is handled.

meldano, 2018-08-24:
I am a Little bit confused…

After many try and error, this solution work

`uint8_t u8x8_byte_stm32_hal_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	switch (msg)
	{
		
		case U8X8_MSG_BYTE_SEND:
		HAL_SPI_Transmit(&SPI_HANDLER, arg_ptr, arg_int, 100);
		while (HAL_SPI_GetState(&SPI_HANDLER) != HAL_SPI_STATE_READY) ; 
		break;
		
		case U8X8_MSG_BYTE_SET_DC :
		if(arg_int)
				HAL_GPIO_WritePin(DISPLAY_DC_GPIO_Port, DISPLAY_DC_Pin, GPIO_PIN_SET);
		else
			HAL_GPIO_WritePin(DISPLAY_DC_GPIO_Port, DISPLAY_DC_Pin, GPIO_PIN_RESET);
		break;
				
		case U8X8_MSG_BYTE_START_TRANSFER:
		HAL_GPIO_WritePin(DISPLAY_CS_GPIO_Port, DISPLAY_CS_Pin, GPIO_PIN_RESET);
		__asm volatile("nop");
		break;
		
		case U8X8_MSG_BYTE_END_TRANSFER:
		HAL_GPIO_WritePin(DISPLAY_CS_GPIO_Port, DISPLAY_CS_Pin, GPIO_PIN_SET);
		__asm volatile("nop");
		break;
		
		default:
		return 0;
	}
	return 1;
}

uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	switch (msg)
	{
		
	case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
		Display_Io_Init();
	break;		
		
	case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second	
		delay_ns(arg_int);	
		break;    
		
	case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
		delay_ns(arg_int * 100);
		break;
		
	case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
		delay_ns(arg_int * 10000);
		break;
		
	case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
		HAL_Delay(arg_int);
		break;

		
//		case U8X8_MSG_GPIO_DC:			// Reset pin: Output level in arg_int
//			if(arg_int)
//				HAL_GPIO_WritePin(DISPLAY_DC_GPIO_Port, DISPLAY_DC_Pin, GPIO_PIN_SET);
//			else
//				HAL_GPIO_WritePin(DISPLAY_DC_GPIO_Port, DISPLAY_DC_Pin, GPIO_PIN_RESET);
//		break;
		
		case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
			if(arg_int)
				HAL_GPIO_WritePin(DISPLAY_RESET_GPIO_Port, DISPLAY_RESET_Pin, GPIO_PIN_SET);
			else
				HAL_GPIO_WritePin(DISPLAY_RESET_GPIO_Port, DISPLAY_RESET_Pin, GPIO_PIN_RESET);
		break;
	}

	return 1; // command processed successfully.
}`

If I use DC Pin in 

`uint8_t u8g2_gpio_and_delay_stm32`

and comment out the DC Pin handling in 

`uint8_t u8x8_byte_stm32_hal_hw_spi`

The lib didn´t work.

And I didn´t implement the CS Pin in the gpio and delay procedure.

olikraus, 2018-08-24:
sure, it is ok to handle the CS/DC line in the byte procedure.

So this issue can be closed?

meldano, 2018-08-24:
Yes, Thank you very much!

closed 2018-08-24 by meldano