olikraus/u8g2 #684: support for SSD1327 128x128 1.5" IC2

ghmac, 2018-08-20:
Hi olikraus,

I just purchased the I2C version of the SSD1327 128x128 1.5" 16-bit grey level OLED by MakerHawk-US.
I'm pretty sure this is same one here: http://www.raspberrypiwiki.com/index.php/1.5_inch_OLED_Shield.
Would you please consider updating your constructor list for these?

Thanks

olikraus, 2018-08-20:
You closed this. So this is not required any more?

ghmac, 2018-08-20:
on second thought I have installed latest library for UNO and do not see it listed in the constructors

olikraus, 2018-08-20:
In the .h file?

ghmac, 2018-08-20:
OK seems this constructor works with most the test sketches:
U8G2_SSD1327_MIDAS_128X128_2_HW_I2C u8g2(U8G2_R0, (/* reset=*/ U8X8_PIN_NONE, (/* clock=*/ SCL, /* data=*/ SDA)));  // page buffer, size = 256 bytes
Full Buffer nothing works. Runs out of memory.
I hope I wrote it well enough that it's correct?
Thanks

olikraus, 2018-08-20:
Out of memory?
Usually the linker garbage collector should be activated to ensure that all is unused blocks are removed.

olikraus, 2018-08-21:
Which uC do you use? 128x128 will require 2KBytes: This will be too much for any ATMega3xx based system like Uno, Mini, Nano or similar.

ghmac, 2018-08-21:
Ahh OK that's why then. I was using a Nano and then UNO. 
Both gave out of memory warnings.
Least it works under Page.
Thanks for all your work and help!

olikraus, 2018-08-21:
:)

closed 2018-08-21 by olikraus