olikraus/u8g2 #685: LCD 128X64 ST7920 - Bad highliting text with setdrawcolor

dromega75, 2018-08-20:
Hi, i have my arduino mini pro with a LCD 128x64 (ST7920), set the u8g2_font_profont11_tr font and the setdrawcolor to 0.
The display show a bad highlight text. 
Here the code

LCD.firstPage();
  do {
  LCD.setFontMode(0);
  LCD.setDrawColor(1);
  LCD.drawStr(0,24,"THERMOHYGROMETER");
  LCD.setDrawColor(0);
  LCD.drawStr(0,34,"THERMOHYGROMETER");
  LCD.drawStr(0,44,"Thermohygrometer");

  } while ( LCD.nextPage() );

[image]

olikraus, 2018-08-20:
Use a font with m or h as last to last letter

olikraus, 2018-08-24:
no feedback.. closing

closed 2018-08-24 by olikraus