olikraus/u8g2 #689: HX1230 Power Save

maopiccoto, 2018-08-27:
I've achieved lower power consumption with **HX1230** display modifying  the PowerSave sequence in this way:
```

static const uint8_t u8x8_d_hx1230_96x68_powersave1_seq[] = {
 U8X8_START_TRANSFER(),         /* enable chip, delay is part of the transfer start */             
  U8X8_C(0x0ae),             /* display off */
 U8X8_C(0x0a5),		/* All pixels on = powersave
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

```
As the Datasheet says:
"If the display all points ON command is executed when the display is in display OFF mode,
power saver mode is entered"

Power consumption falls down 0,01mA

v1993, 2018-08-27:
Why don't make pull request for it?

closed 2018-08-27 by maopiccoto

olikraus, 2018-08-27:
yes, pull request would be great :-)

maopiccoto, 2018-08-29:
Pull request done. Thanks.

olikraus, 2018-08-30:
Merged, thanks a lot!

olikraus, 2018-08-30:
Pull Request: #692