olikraus/u8g2 #690: u8g2_SetupBitmap implementation - or other way to implement pixel scrolling

ulno, 2018-08-28:
I am the author of an Arduino-based IoT framework and currently using u8g2 to implement its display drivers. I recently switched it from a micropython based firmware to C++ and therefore want to replicate the same behavior as before.

In my old framework I could scroll the content of a display, so I could mix graphics and text and still log into the small display moving existing graphics up. As there is no scrolling in u8g2, I was thinking to use another bitmap buffer and then render always this buffer and implement pixel scrolling myself. So I was considering using U8G2_BITMAP for this purpose - however the implementation of u8g2_SetupBitmap is gone. There are prototypes in the header which led me to believe it existed, however I can't find its implementation.

Is there any other way to implement pixel-based scrolling? Should I manage my own bitmap (then it's just sad that I lose all the font management) or is there a safe way to use getBufferPtr and then implement scrolling on that with memmove?

ulno

olikraus, 2018-08-29:
You mean this? https://github.com/olikraus/u8g2/blob/master/sys/bitmap/common/u8x8_d_bitmap.c

> Is there any other way to implement pixel-based scrolling? 

Why do you need the bitmap device? Scrolling is nothing else than redrawing with a different x or y offset.
Example is here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/Shennong/Shennong.ino

ulno, 2018-08-29:
> You mean this? https://github.com/olikraus/u8g2/blob/master/sys/bitmap/common/u8x8_d_bitmap.c

Yes, thanks! No idea, why neither I nor platformio couldn't find it before.

I have seen the shennong example, but as stated before I want to be able to scroll existing content (both drawings and text) that have been drawn into some kind of buffer in advance by the users who remotely (due to given network commands) send drawing commands to this display.

But maybe I have to rethink how a generic display should be best controllable via the network.

olikraus, 2018-08-30:
Ok, I roughly understand your idea. The problem is, that this is not really supported by U8g2.

ulno, 2018-08-30:
Would it make sense to implement that for the full buffer devices? If you have the full buffer of all pixels being drawn (even in tiles), it should be pretty easy to shift these bytes around to scroll by a line, shouldn't it?

Or at least there could be a get_pixel and set_pixel (I know drawPixel exists) for these type of buffers, which would allow the user to do some simple  inefficient implementation themselves.

Please feel free to flag this issue just as a feature request or close it and then ignore it. I have retired all graphics functionality requirements for now from my interface and just draw characters and keep them in a buffer (here is my implementation in my display-device: https://github.com/ulno/ulnoiot-tng/blob/master/lib/node_types/esp8266/src/display.h and https://github.com/ulno/ulnoiot-tng/blob/master/lib/node_types/esp8266/src/display.cpp) - very similar to the log device, I might have replicated some functionality here. So, I might not need that feature for a while either.

olikraus, 2018-08-30:
NOTE 1:

u8x8_d_bitmap will not really fit:
1) It uses two bitmaps
2) It uses malloc()

There is no real need for the two bitmaps, but it simplifies the structure of u8x8_d_bitmap a lot. In fact the architecture of u8g2 assumes two bitmaps: One is on memory and one is on the display controller.
Because u8x8_d_bitmap does not have a display controller connected, it needs to simulate the RAM of that display controller.

NOTE 2:
The assumption is, that you want to reuse the u8g2 graphics procedures. If these procedures are not needed, then an implementation would be simple:
Just create a XBM buffer in RAM and modify the same with a self-written set pixel procedure. Then use u8g2 draw XBM procedure to draw this buffer at different positions.

But I guess, this is not what you want, because you will end up in writing another set of graphics procedures.

NOTE 3:
The correct solutions should have the following properties:
a) The bitmap should have a XBM structure, so that the u8g2 draw XBM procedure can be used to put the bitmap on any other display.
b) There should be no malloc()
c) There should be only one buffer. 
d) Existing u8g2 should be usable to draw something into the bitmap.
Outline of a solution:
Key part of the solution is to create a new low level line draw procedure, which directly writes to the bitmap, bypassing the device sub system: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L289

u8x8_d_bitmap.c mimics a normal device subsystem, bit with this solution, no device subsystem is required. This has to important implications: (A) there is no u8x8 API and (B) we need a complete new setup/init procedure.

NOTE 4:
> Would it make sense to implement that for the full buffer devices? If you have the full buffer of all pixels being drawn (even in tiles), it should be pretty easy to shift these bytes around to scroll by a line, shouldn't it?

There is a u8g2 buffer interface (https://github.com/olikraus/u8g2/wiki/u8g2reference#direct-access-buffer-api). Obviously you could do the shifting directly in that buffer. But it will be difficult to shift in new content.

ulno, 2018-08-30:
Thanks for the insights.

NOTE1: true

NOTE2: yes, I wanted to re-use graphic procedures and not do my own.

I think I will revisit NOTE 4 (looked at that already but haven't totally understood the buffer structure yet) at a later point and try to shape NOTE 3 as something a student could take a  look at and open it as a feature request on my own project.

olikraus, 2018-08-30:
The width must also be a multiple of 8. But this is a XBM limitation.

olikraus, 2018-08-30:
ok, more values need to be set:
```
  u8g2_bitmap.width = u8g2_bitmap.pixel_buf_width;
  u8g2_bitmap.height = u8g2_bitmap.pixel_buf_height;
  
  u8g2_bitmap.user_x0 = 0;
  u8g2_bitmap.user_x1 = u8g2_bitmap.pixel_buf_width;
  u8g2_bitmap.user_y0 = u8g2_bitmap.buf_y0;
  u8g2_bitmap.user_y1 = u8g2_bitmap.buf_y1;  

```

olikraus, 2018-08-30:
ok, it works.

case study is here:

https://github.com/olikraus/u8g2/blob/master/sys/sdl/xbm2/main.c

closed 2018-10-28 by olikraus