olikraus/u8g2 #691: setFontMode with the u8g2.print

303055, 2018-08-29:
Hi olikraus

https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode

How can we use the setFontMode with the u8g2.print ()?
//
u8g2.setFontMode(0);
u8g2.setDrawColor(1);
u8g2.print(a);
//
Not working 

Thx

303055, 2018-08-30:
Hi,

I can not use the clear buffer because I have another function that works at the same time.
I try not settings but I start to desperate ...

I use the full graphic mode:
U8G2_ST7920_128X64_F_SW_SPI u8g2 (U8G2_R0, 13, 7, 6, 8);

I try with a:
u8g2.setFontMode (0); u8g2.setDrawColor (1); u8g2.drawStr (11,23, "");

before :
u8g2.setFontMode (0); u8g2.setDrawColor (1); u8g2.setCursor (11,23); u8g2.print (a);

without success :/

an idea ?

Thank you in advance

303055, 2018-08-31:
Hi olikraus,

Yes of course :
(PZEM004T is used to measure the current )

`#include <U8g2lib.h>
#include <SPI.h>
#include <PZEM004T.h>
#include <Wire.h> 

PZEM004T pzem(10,11);  // (RX,TX) connect to TX,RX of PZEM
IPAddress ip(192,168,1,1);
  
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 7, 6, 8);

void setup() {
pzem.setAddress(ip);
Serial.begin(9600);
u8g2.begin();  
}

void loop() {

float p = pzem.power(ip);

u8g2.setFont(u8g2_font_crox5hb_tf );
u8g2.setCursor(11,23); u8g2.print("      ");
u8g2.sendBuffer();
u8g2.setCursor(11,23); u8g2.print(p,0);
u8g2.sendBuffer();
Serial.println(p,0);
delay(10);

}`
the video : 
https://youtu.be/Zy8QuRItAks

thx in advance

olikraus, 2018-08-31:
ok, but what is the problem with the font?

I mean, there are some other problems, like you do not clear the screen and the buffer is send twice (which is useless)..

A better version would be this:
```
void loop() {

float p = pzem.power(ip);

u8g2.clearBuffer();
u8g2.setFont(u8g2_font_crox5hb_tf );
u8g2.setCursor(11,23); u8g2.print(" ");
u8g2.setCursor(11,23); u8g2.print(p,0);
u8g2.sendBuffer();
Serial.println(p,0);
delay(100);

}
```

303055, 2018-08-31:
Hi

Now it work! 

It was a bad understanding of the bookstore on my part
I thought it was necessary to send after clear:
u8g2.clearBuffer ();
u8g2.sendBuffer ();

but no!
just in the loop:
u8g2.clearBuffer ();
// write ...
u8g2.sendBuffer ();

Sorry to annoy you for a bad read ..

Thank you for all your work

And thank you for answering me

olikraus, 2018-08-31:
:)