olikraus/u8g2 #695: Overflow of x-values

fredpcar, 2018-09-03:
I am using the following code snippet to experiment a bit with my e-paper display:

        U8G2_IL3820_296X128_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 9, 8);
	u8g2.setDrawColor(1);
	u8g2.firstPage();
	do
	{
		int y = 64;
		for (int i=0; i < 296; i++){
			  y = retrieveByte(i)*5;
			  u8g2.drawVLine(i, y, u8g2.getHeight()-20 -y);

		   }
		u8g2.setFont(u8g2_font_timB10_tf);
		u8g2.drawStr(260, 127,"empty");
	} while (u8g2.nextPage());

So this is supposed to draw a graph across the entire display width but there seems to be an overflow at 255/255 of the x-parameter - as if it would be a single byte !?? Means my graph after x-position 256 and my text positioned at 260 are shown starting at the most left (low x) position.
I strongly suspect my own fault but cannot find it.

JamesGKent, 2018-09-03:
you need to enable [16 bit mode](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#16-bit-mode) on larger displays

fredpcar, 2018-09-03:
And that was it of course - thank you so much for the speedy resolution. Closing this.

closed 2018-09-03 by fredpcar