olikraus/u8g2 #697: Q: Fastest supported 128x64 display chip?

Triangulix, 2018-09-05:
Hi
I'm currently using a 128x64 pixel display with ST7920. The transmission of the screen buffer content to the display via SPI takes 28ms. This blocks my main loop for too long.
Unfortunately I couldn't find an overview over the transfer speeds u8g2 supports with the supported chips.
Which display chip is currently the fastest supported solution for a monochrome 128x64 display?

Thanks

olikraus, 2018-09-05:
For any other chip will SPI will be faster. The problem with the ST7920 is the fact, that it requires two SPI bits to transfer one monochrome pixel value.
For a controller like the ST7565, one bit on the SPI bus is used for the color of one pixel (twice the speed compared to the ST7920).

Triangulix, 2018-09-06:
Thanks, that is great to know.
Then I will try a display with ST7565R or KS0107.

Triangulix, 2018-10-14:
Hi
I'm sorry but I have to reopen this topic.
In the meantime I have ordered a new display, this time with KS0107+KS0108 chip, made a new cable and just tested it. The transmission of the frame buffer to the display takes now 42ms instead of 28ms so it is even slower than before. :-(
The ST7920 was connected via SPI, the KS0108 is now connected via parallel interface. How can this be slower?

Thanks

olikraus, 2018-10-14:
Yes. Parallel Interface is even slower.
I think Bill's openGLCD is optimized for ks0108.

Triangulix, 2018-10-14:
Which combination of chip and interface would you recommend for 128x64 or 196x64 pixel and minimal blocking time in the main loop?

Triangulix, 2018-10-14:
Is there any way to break the data transfer up into chunks so that the main loop is not blocked for such a long time?

olikraus, 2018-10-14:
With the picture loop approach you can execute the  loop body in parallel to other tasks.

There are many other interface chips available: st75256, st7665, uc16xx

Triangulix, 2018-10-14:
Which of these other chips allows for the fastest data transmission from Arduino Due to the display?

olikraus, 2018-10-14:
I put together some measurements in FPS.ino page buffer example.

Triangulix, 2018-10-15:
Oh, I didn't know that there already is a performance measurement tool. I was about to suggest to make something available so that users can measure the performance of their display/interface/Arduino combination and build up a public performance chart.

closed 2018-10-28 by olikraus