olikraus/u8g2 #699: I2C not working on SSD1327

Andrzej1000, 2018-09-07:
Hi.
I have SSD1327 form waveshare.
The library working on SPI.
Unfortunately I2C does not work.
The display does not light anything.
I checked example code from waveshare and it works.
Olikraus could you check what's going on?
Thanks...

olikraus, 2018-09-07:
You say, SPI works, but I2C doesn't work, right?

Andrzej1000, 2018-09-08:
Yes.
SPI working but I2C not work.

Andrzej1000, 2018-09-08:
SPI working with your library:
//U8G2_SSD1327_MIDAS_128X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SSD1327_MIDAS_128X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
//U8G2_SSD1327_EA_W128128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

I2C with library U8G2 not working:
//U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2018-09-08:
ok. Did you connect DC input line of the display to GND (so that the default I2C address is 0x78)?

olikraus, 2018-09-08:
I crosschecked the software. Everything should be ok.

Andrzej1000, 2018-09-08:
OK. It works. DC connected to GND. Thank you very much !!!
Please, make init string U8G2 for the default address without this connecting...?

olikraus, 2018-09-08:
In I2C mode, you must connect connect the DC input to either low or high, you can not leave it open. If open the display will draw much more current and the I2C address is unpredictable. All in all you should follow the wiring instructions in chapter 8 of the SSD1327 controller.
Depending on the level applied to DC input,  you should inform U8g2 about the selected I2C address.

See here:
https://electronics.stackexchange.com/questions/7179/is-it-really-a-bad-idea-to-leave-an-mcu-input-pin-floating
and here:
https://www.quora.com/Why-is-it-not-advised-to-float-inputs-to-CMOS-gates-under-any-circumstances#

olikraus, 2018-09-08:
:-)

s1sm1x, 2019-06-05:
Hi olikraus,

Still on this issue, I'm having a lot of trouble on gettin the 128x128 SD1327 1.5' module (waveshare) to work. After reading every post about SD1327 written from you, I still can't display as I want.
I'm using a nodemcu compatible board: Wemos D1 mini.
I use this code to test several font sizes:

#include <Arduino.h>
#include <U8g2lib.h> 
#include <Wire.h>

/*
  U8g2lib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.

  This is a frame buffer example.
*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected
//U8G2_NULL u8g2(U8G2_R0);  // null device, a 8x8 pixel display which does nothing
//U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ D1, /* data=*/ D2, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

// End of constructor list

#define MINI_LOGO

void setup(void) {
  //pinMode(9, OUTPUT);
  //digitalWrite(9, 0); // default output in I2C mode for the SSD1306 test shield: set the i2c adr to 0
  Serial.begin(115200);

  u8g2.begin();
}

void draw1(void)
{
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb46_mf);
  u8g2.drawStr(0, 63, "1111");
  u8g2.sendBuffer();
  Serial.println("46");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb49_mf);
  u8g2.drawStr(0, 63, "1111");
  u8g2.sendBuffer();
  Serial.println("49");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb53_mf);
  u8g2.drawStr(0, 63, "1111");
  u8g2.sendBuffer();
  Serial.println("53");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb57_mn);
  u8g2.drawStr(0, 63, "1111");
  u8g2.sendBuffer();
  Serial.println("57");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb63_mn);
  u8g2.drawStr(0, 63, "1111");
  u8g2.sendBuffer();
  Serial.println("63");
  delay(1000);
}
void loop(void) {

  drawLogo();
  u8g2.sendBuffer();
  draw1();

  delay(1000);
}

But at the end, I only get a small portion of the display to work horizontally at the top like the : 
[image]
[image]
[image]

The "1111" string seems to be printed, but the numbers don't show on the whole screen.
The final idea is to display time ("12:12")  as big as it can, That's why I chose a 1.5' display instead of the available 0.91' ou 0.96'. 

Could you help me? 
This is for a direct application project and I'm have been freaking out with his type of oleds. But, at this time, I feel that I'm pretty close to a working solution!

Thank you for the paid attention,

Pedro Oliveira / s1sm1x

olikraus, 2019-06-05:
Use the `_F_` constructor: `U8G2_SSD1327_EA_W128128_F_SW_I2C `, see also here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#buffer-size
You must not use sendBuffer together with the `_1_` constructor.

olikraus, 2019-06-05:
Or here: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L134

s1sm1x, 2019-06-05:
> Use the `_F_` constructor: `U8G2_SSD1327_EA_W128128_F_SW_I2C `, see also here: https://g

#include <Arduino.h>
#include <U8g2lib.h>

#include <Wire.h>

//U8G2_NULL u8g2(U8G2_R0);  // null device, a 8x8 pixel display which does nothing
//U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

//U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ D1, /* data=*/ D2, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_2ND_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//after olikraus
U8G2_SSD1327_EA_W128128_F_SW_I2C  u8g2(U8G2_R0, /* clock=*/ D1, /* data=*/ D2, /* reset=*/ U8X8_PIN_NONE);

// End of constructor list

#define MINI_LOGO

void setup(void) {
  //pinMode(9, OUTPUT);
  //digitalWrite(9, 0); // default output in I2C mode for the SSD1306 test shield: set the i2c adr to 0
  Serial.begin(115200);

 
  u8g2.begin();
  
}
void draw1(void)
{
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(0,20,"Hello World!");
  u8g2.sendBuffer();
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb46_mf);
  u8g2.drawStr(50, 50, "1111");
  u8g2.sendBuffer();
  Serial.println("46");
  delay(2000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb49_mf);
  u8g2.drawStr(50, 50, "1111");
  u8g2.sendBuffer();
  Serial.println("49");
  delay(2000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb53_mf);
  u8g2.drawStr(50, 50, "1111");
  u8g2.sendBuffer();
  Serial.println("53");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb57_mn);
  u8g2.drawStr(50, 50, "1111");
  u8g2.sendBuffer();
  Serial.println("57");
  delay(1000);
  u8g2.clearBuffer();
  u8g2.setFontDirection(0);
  u8g2.setFont(u8g2_font_inb63_mn);
  u8g2.drawStr(50, 50, "1111");
  u8g2.sendBuffer();
  Serial.println("63");
  delay(1000);
}
void loop(void) {
  // u8g2.clearBuffer();
  drawLogo();
  u8g2.sendBuffer();
  draw1();
  //u8g2.sendBuffer();
  delay(1000);
}

It partially write the logo, but it reset through Soft wdt with the message code:

Soft WDT reset

ctx: cont 
sp: 3ffffbe0 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffd90:  00000078 3fffff30 0000001c 00000030  
3ffffda0:  40106538 00000001 00000040 feefeffe  
3ffffdb0:  40106538 00000001 00000040 40214f9a  
3ffffdc0:  40106538 00000001 00000040 40214f9a  
3ffffdd0:  40106538 00000001 00000040 40214f9a  
3ffffde0:  40106538 00000001 00000040 40214f9a  
3ffffdf0:  40106538 00000001 000000ff 40106569  
3ffffe00:  40106538 00000001 000000ff 40214f9a  
3ffffe10:  40106538 00000000 00000005 40214f9a  
3ffffe20:  00000091 00000001 00000000 00000008  
3ffffe30:  000000bc 000000ff 3ffee900 40214a82  
3ffffe40:  000000bc 000000ff 3ffee900 402019f2  
3ffffe50:  000000bc 000000ff 3ffee900 40201a14  
3ffffe60:  000000bc 00000000 3ffee900 40201a5b  
3ffffe70:  3fffff50 3ffef1a4 3ffee900 40201aa4  
3ffffe80:  00000018 3ffef1a8 3ffee900 40201b59  
3ffffe90:  00000018 3ffef1a4 3ffee900 4021502b  
3ffffea0:  3ca03c0b 000000af 3ffee900 40201c23  
3ffffeb0:  00000018 00000020 3ffee900 40201d25  
3ffffec0:  3ffef1a4 00000000 00000000 000000fb  
3ffffed0:  3fffff50 00000001 3ffee900 402149a9  
3ffffee0:  00000000 3ffee900 3ffee96c 40201e8e  
3ffffef0:  0000000e 00000000 00000010 3ffeeab4  
3fffff00:  00000000 0000001a 3ffee900 40201ef8  
3fffff10:  8fae16b2 0000000a 00000064 00000001  
3fffff20:  00000002 0000000f 3ffee900 40201ef8  
3fffff30:  3ffef124 000f0010 3ffe8a98 00000000  
3fffff40:  00000002 00000002 3ffee900 40214a08  
3fffff50:  3ffeeaa4 00020010 3ffee900 00000000  
3fffff60:  20a715ec 0000000b 3ffee900 40201060  
3fffff70:  00000010 00000010 00000001 402147a2  
3fffff80:  00000000 3ffee8f8 3ffee900 3ffef374  
3fffff90:  3fffdad0 00000000 3ffee900 402010bf  
3fffffa0:  3fffdad0 00000000 3ffef36c 40214e8b  
3fffffb0:  3fffdad0 00000000 3ffef36c 40215814  
3fffffc0:  feefeffe feefeffe 3ffe85dc 401000e5  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

infinite loop inside some draw?
btw it works pretty good on a 1.3' with SH1106 driver when the constructor changes.

I'm stuck on this. Can you help me pls?

Thank you for your energy and availability,

Pedro Oliveira

s1sm1x, 2019-06-05:
> Or here: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L134

yes, I understood. Now , I'm using what you said. _F_ version

thank you

s1sm1x, 2019-06-05:
> #include <Arduino.h>
> #include <U8g2lib.h>
>
> #include <Wire.h>
>
> //U8G2_NULL u8g2(U8G2_R0); // null device, a 8x8 pixel display which does nothing
> //U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* [...]

[image]

That's how it looks like.

olikraus, 2019-06-05:
I do not know your environment, but I would say, that this is not a u8g2 problem. You probably need to increase WDT on your environment.

See here: https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

s1sm1x, 2019-06-05:
> I do not know your environment, but I would say, that this is not a u8g2 problem. You probably need to increase WDT on your environment.
> 
> See here: https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/
If I use this constructor, it crashes after drawing the logo, but before drawing the strings:
U8G2_SSD1327_EA_W128128_F_SW_I2C  u8g2(U8G2_R0, /* clock=*/ D1, /* data=*/ D2, /* reset=*/ U8X8_PIN_NONE);

if I use this: U8G2_SSD1327_EA_W128128_F_HW_I2C  u8g2(U8G2_R0, /* clock=*/ D1, /* data=*/ D2, /* reset=*/ U8X8_PIN_NONE);
It stops the writing but doesn't reset.

Does this give you any further information?

Thank you in advance,

Pedro Oliveira

olikraus, 2019-06-05:
Looks like a wdt problem...