olikraus/u8g2 #702: ssd1306 help

mjphillips1981, 2018-09-11:
I was successful at getting a 128x32 ssd1306 via i2c to work but I am now trying to change the code to work with a 128x64 ssd1306 and cannot get it to display anything.  My original display was the Adafruit OLED feather and the new one is the Adafruit 326.  I have setup the new display to work with I2C as described in the Adafruit instructions.  Any help would be appreciated.

I changed my setup from 
       u8g2_Setup_ssd1306_i2c_128x32_univision_f 
          to
       u8g2_Setup_ssd1306_i2c_128x64_noname_f

olikraus, 2018-09-11:
You use an Arduino IDE?

It might be worth to try sw emulated I2C.

mjphillips1981, 2018-09-11:
No using c on an Atmel samg55.  Software I2C. Like i i said everything works great on the 128x32 but not on the 128x64

olikraus, 2018-09-12:
Well, not sure how I can help here. I think this is a connection problem. Maybe also some pin assignment is wrong or you need to update the gpio&delay procedure.

mjphillips1981, 2018-09-12:
I got it to work.  I initially did not have the reset pin controlled by the micro so after adding the reset control everything works as it should.

closed 2018-09-12 by mjphillips1981