olikraus/u8g2 #703: Crox family fonts - how to print in Russian

zhenek-kreker, 2018-09-12:
Crox family is marked as a Cyrillic fonts, but I cannot use print() function with Cyrillic text. With enableUTF8Print() There's just a couple of spaces instead of the text showing up on the screen. Disabling doesn't help as well, just instead of spaces weird glyphs appear.

The fonts are very nice, I'd very much like to use them.

v1993, 2018-09-12:
They are *not* unicode fonts, you should use char codes form the left.
For example, Т (большая, руссакя) is `\xd2`(`00d0` + third in row).

zhenek-kreker, 2018-09-12:
So there's no easy way to use print? Like
`u8g2.print("Капюшон");`
?

v1993, 2018-09-12:
I didn't found one :disappointed:.

You can try writing some preprocessors or scripts to don't do conversion manually.

olikraus, 2018-09-12:
Indeed the crox fonts are encoded in CP-1251, so UTF-8 does not make sense.
(Reference: http://www.kovrik.com/sib/russify/x-windows/#win-fonts)

The main idea is this: If you put your Editor into 1251 mode, you can just type 
```
u8g2.print("Капюшон");
```
and U8g2 should print your text exactly like this. In such a case you also must not enable UTF-8.

The problem is more the Arduino IDE: To my knowledge the Arduino IDE only works in UTF-8 mode. Also the compilers will work in UTF-8 mode. At least I do not know whether it is possible to change Arduino IDE here (it might be also impossible).

As a result: The crox fonts do not really make sense at the moment. You can still use the hex notation from above.

Please also note that there are some "real UTF-8" cyrillic fonts. They are usually postfixed with "_cyrillic" or "_all". 
Here are som examples:
```
u8g2_font_6x12_t_cyrillic
u8g2_font_cu12_t_cyrillic
u8g2_font_unifont_t_cyrillic
u8g2_font_haxrcorp4089_t_cyrillic
```
The NBP collection contains also several fonts with cyrillic glyphs: https://github.com/olikraus/u8g2/wiki/fntgrpnbp

My suggestion is to look around in other font groups to see if you find something suitable.

do not have proper UTF-8 encoding. I just included the fonts as they are.

v1993, 2018-09-12:
I suggest using separate file where you have all utf8 strings named like `mystrings.h.in`:

    const char* STR_CAPUSHON = "Капюшон"
    ...
Then use tool like `enconv` to convert:

    cat mystrings.h.in | enconv -x CP-1251 > mystrings.h
And include `mystrings.h` in your `.cpp` file. Don't forget re-run converter after changing `mystrings.h.in`!

zhenek-kreker, 2018-09-18:
Thanks! I've finally turned to `u8g2_font_pxplusibmvga8_m_all` font, it looks very nice.

closed 2018-09-18 by zhenek-kreker

olikraus, 2018-09-18:
:-)