olikraus/u8g2 #704: (Own) 7segment font [NotWorking]

MrAntza99, 2018-09-13:
Hi, I have been tried make own font two days and I made it but cant understand how to apply it in arduino?

Also I tried premade font what you have on library but its not working 
u8g2.setFont(u8g2_font_7Segments_26x42_mn);
^this not work, why? it doenst recognize it.

I made font with Fony, I have BFF-file etc.
I tried convert it to .c code but cant do it.

Any simple help? Something little what I just forgot there?

olikraus, 2018-09-13:
> Also I tried premade font what you have on library but its not working
u8g2.setFont(u8g2_font_7Segments_26x42_mn);
^this not work, why? it doenst recognize it.

The font is here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L1224

What exactly is the error message? Do you use the latest u8g2 version?

> Any simple help? 
Here: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L204

MrAntza99, 2018-09-13:
Ahh.. U8G2 needed updates etc..

BUT I dont have any knowledge still how to add that own font, I tried launch yours bdfconv.exe but it just closed itselfs in same moment.. How to use this code? 

Q: Which commandline options are required for bdfconv?
A: "bdfconv -f 1 -m '32-255' -n fontname -o myfont.c myfont.bdf"
"-f 1" generates a u8g2 font.
"-m '32-255'" selects unicode 32 to 255. On Windows, please use double quotes: -m "32-255"
"-n fontname": This is the name of the font in C/C++/Ino files.
"-o myfont.c": The font array will be stored in this file.
"myfont.bdf": The input file with the font data (bdf format).

olikraus, 2018-09-13:
bdfconv is a commandline tool. You need to execute bdfconv.exe from "cmd.exe": You need to start cmd.exe first.

Maybe this helps:

http://www.codejacked.com/a-beginners-guide-to-the-command-prompt/

Once you are able to work with the command prompt, you can start bdfconv with the mentioned arguments. This will finally generate a file (myfont.c in the example above) which has to be pasted into your .ino file.

olikraus, 2018-10-28:
feel free to reopen, if required...

closed 2018-10-28 by olikraus