olikraus/u8g2 #707: NHD-2.7-12864WDY3

steronydh, 2018-09-19:
I've been using u8glib and M2tklib on an NHD‐2.7‐12864UCY3 with SSD1325, but this part is now discontinued. The replacement NHD-2.7-12864WDY3 has an SSD1322, which u8g2 supports, but M2tklib doesn't work with u8g2.

What is the recommended migration pathway? Is there something like M2tklib that will work with u8g2?

olikraus, 2018-09-19:
Well... no, M2tk is not ported.
I once started, but didn't found further time.

Current state is here: https://github.com/olikraus/m2tklib

steronydh, 2018-09-19:
That's a shame, M2tk is really useful. Would it take much to convert it to u8g2, or are there similar alternatives with the same capabilities? I'd rather not have to write my own unless I have to!

olikraus, 2018-09-19:
Well... it is a question of time... job... family there is not much left im my spare time.
I also felt, that M2tk is too complicated for the average Arduino user. I think you are one of the rare users who went into the details of M2tklib. 

Actually it shouldn't be that complicated. I just would need to follow the u8g to u8g2 porting guide.
I just didn't do it until now. :-(

olikraus, 2018-09-19:
The graphics part is already isolated in M2tk and I also started the porting:
https://github.com/olikraus/m2tklib/tree/master/dev/u8g2

I do not remember the state, but maybe it is already partly working... 
I try to get some time today evening to check the current state. Maybe it is already sufficient for you.

olikraus, 2018-09-19:
well... I made some updates here and there, but still a long way to go...

steronydh, 2018-09-20:
Thanks for looking at it but if there really are very few users please don't put a lot of time into it, you've got much more productive projects on the go!

closed 2022-02-09 by kravlost