olikraus/u8g2 #709: SSD1325 No response

Tulsa2008, 2018-09-20:
I hate to waist your time but I am having a problem You may be able to help me solve very quickly.
I am using a 2.7" GLED display (ADAFruit # 2764) using a SSD1325 controller.  The processor I am using is an Arduino Mega2560.  I am using the hardware SPI interface.  

Since the display operates on 3 volts, I am using a CD 4050 interface chip to reduce the drive voltage fro 5 volts to 3 volts.

The display works using the ADAfruit Lib.  Although I am having noise problems

I am using your constructor:
   U8G2_SSD1325_NHD_128X64_F_4W_HW_SPI u8g2(U8G2_R0, 48, 49, 47);

When I use your lib, the display stays blank.  I have tried both the HW and SW versions of the constructor.  

I have used your Lib with a 7565 display and the lib works great.  I would really like to use your lib over the ADA lib.  
The code I am using is as follows:
<
#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

/* Constructor */
//U8G2_SSD1325_NHD_128X64_F_4W_HW_SPI(U8G2_R0, cs, dc, reset)
//  U8G2_SSD1325_NHD_128X64_F_4W_SW_SPI u8g2(U8G2_R0,7, 6, 48, 49, 47);
  U8G2_SSD1325_NHD_128X64_F_4W_HW_SPI u8g2(U8G2_R0, 48, 49, 47);

    byte i = 0;

void setup(void) {
  Serial.begin(115200);
  u8g2.begin();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  Serial.println("Boot Complete");
}

void loop(void) {
    i = i +1;
    u8g2.clearBuffer();          
    delay(1000);
    u8g2.setFont(u8g2_font_5x7_tr);  // Small Font 
//    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,20,"Hello World!");
    u8g2.drawStr(10,30,"i = "); u8g2.drawStr(0,30,i);
    delay(5000);
}
>
Any light you can shed will be greatly appreciated.
Ken

Tulsa2008, 2018-09-20:
Thanks for the quick answer. 

I don't know how I missed that one??  I am using your lib for a system using 7576 displays and I see in the code for them the "u8g2.sendBuffer();" command. 

Thanks again for the kick in the pants.
Ken

closed 2018-09-20 by Tulsa2008

olikraus, 2018-09-21:
:)