olikraus/u8g2 #712: Support Max64x8.

hevet, 2018-09-28:
Is it possible to add support for two 32x8 max modules connected in a 64x8 line?

olikraus, 2018-09-28:
should be no problem...

hevet, 2018-09-28:
Super, thanks.

olikraus, 2018-10-08:
I have created constructor `U8G2_MAX7219_64X8_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* cs=*/ 10, /* dc=*/ U8X8_PIN_NONE, /* reset=*/ U8X8_PIN_NONE);`

Can you test this device? I am also interested whether the setContrast fuction works.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

hevet, 2018-10-09:
Yes, it will test today.

hevet, 2018-10-09:
Everything works fine, contrast also. Thanks for the good job.

olikraus, 2018-10-09:
:)

closed 2018-10-09 by olikraus