olikraus/u8g2 #713: Global memory problems

fmarzocca, 2018-09-30:
Hi Oli,
I am running low global memory problems with your nice library.
From tests I made, the instruction which is eating my memory is this:

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

It eats 1229 bytes of global memory from the 20148 available, leaving very few room for my sketch.
Am I doing something wrong? Is there any way to reduce this global memory consumption?
Thanks
Fabio

v1993, 2018-09-30:
Did you meant 2048 instead of 20148?

v1993, 2018-09-30:
Are you talking about RAM or FLASH? If RAM, try using `U8G2_SSD1306_128X64_NONAME_1_SW_I2C` (or `U8G2_SSD1306_128X64_NONAME_2_SW_I2C`) and page loop.

It can make your program a bit more complicated (and slower), but there is no other way (until you buy some controller with more RAM, like some ESP ones).

v1993, 2018-09-30:
Read more here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#buffer-size

fmarzocca, 2018-09-30:
Hi, I know the Page Loop. But I am getting that great memory occupation without ANY print instruction. Just include the library and set the SSD1306: it eats 1229 bytes over 2048 of global memory! I suppose that using print and fonts (even with the page loop) I will run out of memory soon, not considering my sketch.

fmarzocca, 2018-09-30:
Are you saying that u8g2 is not recommended for Arduino Nano projects?

v1993, 2018-09-30:
Well, do you get it with buffered constructor?

Just think: buffer is allocated when you create object and not when you access it.

Not sure about arduino nano, but works well on Uno. I don't know much about u8x8 mode, but it seem really memory-saving one (but text-only).

fmarzocca, 2018-09-30:
> Well, do you get it with buffered constructor?

What do you mean? I just include the library and execute that line before setup.

fmarzocca, 2018-09-30:
`U8G2_SSD1306_128X64_NONAME_1_SW_I2C` did the trick1 Thank you!

closed 2018-09-30 by fmarzocca

v1993, 2018-09-30:
Well, I'll describe library a bit:

* Full-buffer
    * When you call constructor, full copy of display memory is created
    * When you print/draw, you internal copy is changed
    * When you call `sendBuffer()`, data is flushed to display
* Page-buffer
    * When you call constructor, partial copy of display memory is created
    * In page loop
        * When you print/draw, you internal copy is changed (only correspondenting part)
        * When you call `nextPage()`, data is flushed to display
Hope that this will help you with understanding!

olikraus, 2018-09-30:
:)

fmarzocca, 2018-09-30:
Thank you very much. It is clear!
One last thing: what can I do if I need to fully clear the display in a pagebuffer loop? Is it allowed?

olikraus, 2018-09-30:
There are two options:
- Use an empty pagebuffer loop
- Use clearDisplay() command

v1993, 2018-09-30:
Abother important thing: ALL pageloop iterations MUST draw exactly same pucture!

fmarzocca, 2018-09-30:
Yes, thanks!