olikraus/u8g2 #715: Multiple I2C devices

fmarzocca, 2018-10-02:
Hi,

I am happy with u8g2 library, which I am using in several of my projects. Today I had the need to ad a second I2C slave on the bus (a laser Time of Flight sensor), but when I start it, the display is not showing anything, it is all black.

I have checked the 2 device addresses are different. I am also investingating with the sensor's owner, but for the time being, are you aware of any problems of your library on a multiple I2C device configuration? I am using ARduino Nano and I have also installed 2 pullup resistors of 10K each, on A5 and A4.

olikraus, 2018-10-02:
Maybe check the first question of the FAQ:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

fmarzocca, 2018-10-02:
Hi Oli,

sorry but that first answer is not clear to me. There are two options: 
**(A) use xxx_HW_I2C()**  : 
I am using `U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);`
Do you mean that I can replace _SW with _HW ?
  
**(B) use different pins with xxx_SW_I2C()**
Using different pins than SCL and SDA?

v1993, 2018-10-02:
A) SW = software emulated. HW = hardware supported (built-in I2C controller, not sure about arduino nano).

B) Use different SDA/SCL for your things and u8g2 display.

fmarzocca, 2018-10-02:
By changing _SW with _HW seems that the display goes crazy. Fonts are not rendered as they should and sometimes it is not readable.
See picture:
[image]

olikraus, 2018-10-02:
Well, thiis looks more like problem with the picture loop and not with a I2C problem.
Maybe (for testing) use the F constructor instead:
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0);`

Also: do not add any I2C pin numbers. U8g2 will select the available I2C hardware automatically.

fmarzocca, 2018-10-02:
It seems I have fixed with this:
`
U8G2_SSD1306_128X64_NONAME_2_HW_I2C u8g2(U8G2_R0); `

I cannot use the F_SW constructor as it is eating all the global memory.
Thank you for your kind support

closed 2018-10-02 by fmarzocca

olikraus, 2018-10-03:
:)

fmarzocca, 2018-10-03:
Allow me an off-topic, also to send you my gratitude for your great library; these are 2 projects of mine which use u8g2 and within the article I reported my appreciation for your library:

https://www.hackster.io/fmarzocca/ultrav-a-portable-uv-index-meter-1a686b

https://www.hackster.io/fmarzocca/measurino-a-measuring-wheel-proof-of-concept-aa388f

Thank you again!
Fabio