olikraus/u8g2 #718: Support for ERC12864FS-655

mattogodoy, 2018-10-03:
Hi!
I love your library, thanks for sharing it :)

I am already using U8g2 for a personal project, but I wanted to make the board smaller, so I bought a **ERC12864FS-655** display (https://www.buydisplay.com/default/2-8-inch-128x64-graphic-module-serial-display-spi-lcd-w-optl-touch-screen).

This is the datasheet of the display: https://www.buydisplay.com/download/manual/ERC12864-655_Series_Datasheet.pdf
And this is the datasheet of the IC (**UC1701**): https://www.buydisplay.com/download/ic/UC1701.pdf

The problem is that I can't get it to work. 
I'm using an **ESP32** Doit board and I'm trying to connect the display using **4 wire SPI**. I've tried these setups and some variants:

```
U8G2_UC1701_EA_DOGS102_1_4W_SW_SPI
U8G2_UC1701_MINI12864_1_4W_SW_SPI
U8G2_ST7565_ERC12864_1_4W_SW_SPI
```

So far the one that shows some results is `U8G2_UC1701_EA_DOGS102_1_4W_SW_SPI`.
This is what you can see in the display: https://i.imgur.com/rLWW7nl.jpg
If I restart the ESP32, the display draws "Hello world", but it distorts over time: https://i.imgur.com/FRaOBri.gifv

I also took a look at #177, but no luck.

This is the code I'm using to test:
```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_UC1701_EA_DOGS102_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 12, /* cs=*/ 14, /* dc=*/ 25, /* reset=*/ 27);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

And this is the wiring between the display and the ESP32:

```
LCD                 ESP32
=========================
1	CS0         14
2	RST         27
3	CD          25
4	WR0         GND
5	WR1         GND
6	D0          GND
7	D1          GND
8	D2          GND
9	D3          GND
10	D4          GND
11	D5          GND
12	D6 (SCK)    13
13	D7 (SDA)    12
14	VDD         3v3
15	VSS         GND
16	VB0+        Nothing
17	VB0-        Nothing
18	VB1+        Nothing
19	VB1-        Nothing
20	VLCD        GND
21	BM0         GND
22	BM1         GND
```

There is a big chance I am missing something, but I'm out of ideas.
Is the library compatible with this display?

Thank you!

olikraus, 2018-10-03:
This is not a u8g2 issue.

```
16	VB0+        Nothing
17	VB0-        Nothing
18	VB1+        Nothing
19	VB1-        Nothing
```
You *must* connect one capacitor between VB0+ and VB0- and another capacitor between VB1+ and VB1-. Suggestion is to use two 1uF ceramic caps.

olikraus, 2018-10-03:
Another mistake is the connection of VLCD. It should be connected to GND via a third 1uF ceramic cap.
Please have a look at page 4 for this display: https://www.lcd-module.com/eng/pdf/grafik/dogs102-6e.pdf
This also applies to the ERC display.

margce, 2018-10-03:
Hello Matt,

I use the same display in one of my projects and it works just fine. I used the setup u8g2_Setup_uc1701_mini12864_f to get me started because it's the same UC1701 

Make sure to follow the connection as described by the manufacturer here,
https://www.buydisplay.com/download/interfacing/ERC12864-655_Interfacing.pdf 

I had to tweak the init sequence to work better,

Here is the default values

```
static const uint8_t u8x8_d_uc1701_mini12864_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x040),		                /* set display start line to 0 */
  
  U8X8_C(0x0a0),		                /* ADC set to reverse */
  U8X8_C(0x0c8),		                /* common output mode */
  
  U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
  U8X8_C(0x0a2),		                /* LCD bias 1/9 */
  U8X8_C(0x02f),		                /* all power  control circuits on */
  U8X8_C(0x0f8),		/* set booster ratio to */
  U8X8_C(0x000),		/* 4x */
  U8X8_C(0x023),		/* set V0 voltage resistor ratio to large */
  U8X8_C(0x081),		/* set contrast */
  U8X8_C(0x027),		/* contrast value */
  U8X8_C(0x0ac),		/* indicator */
  //  0x000,		/* disable */
  
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x0a5),		                /* enter powersafe: all pixel on, issue 142 */
  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

```
And I modified as below:
- Added Clear SHL
- Adjusted the voltage resistor divider ( you might have to tweak as well depending on your resistor configuration, this value is based on the values recommended by ERC schematic)
- Adjusted default constrast value

```
static const uint8_t u8x8_d_uc1701_mini12864_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x040),		                /* set display start line to 0 */
  
  U8X8_C(0x0a0),		                /* ADC set to reverse */
  U8X8_C(0x0c0),		                /* clear shl */
  U8X8_C(0x0c8),		                /* common output mode */
  
  U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
  U8X8_C(0x0a2),		                /* LCD bias 1/9 */
  U8X8_C(0x02f),		                /* all power  control circuits on */
  U8X8_C(0x0f8),						/* set booster ratio to */
  U8X8_C(0x000),						/* 4x */
  U8X8_C(0x026),						/* set V0 voltage resistor ratio to large */
  U8X8_C(0x081),						/* set contrast */
  U8X8_C(0x039),						/* contrast value */
  U8X8_C(0x0ac),						/* indicator */
  //  0x000,		/* disable */
  
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x0a5),		                /* enter powersafe: all pixel on, issue 142 */
  
  U8X8_END_TRANSFER(),        	     	/* disable chip */
  U8X8_END()             				/* end of sequence */
};

```

mattogodoy, 2018-10-04:
Great! Thank you both so much.

I'll follow your recommendations and let you know .

mattogodoy, 2018-10-08:
Hi, guys!

The capacitors finally arrived and now it works! 🎉

I ended up using `U8G2_UC1701_MINI12864_1_4W_SW_SPI`, which works perfect but I'll consider make the modifications suggested by @margce, specially for the initial contrast.

Now a final question:
At this point you might already realised I'm no electronics expert. I've been making some tests out of curiosity and I found that adding a capacitor between `VB0+` and `VB0-`, and another between `VB1+` and `VB1-` is enough for the LCD to work correctly.

According to the interfacing document (https://www.buydisplay.com/download/interfacing/ERC12864-655_Interfacing.pdf), I also have to connect a 10uF and 0.1uF capacitors in parallel between `VDD` and `VSS`, but removing them causes no visible effect. No artifacts or anything as far as I can see.
Same thing happens if I remove the 10M resistor and the 0.33uF capacitor connected to `VLCD`. If I leave it unconnected everything still works ok. No apparent difference.

Do you know what are those components for? Can I get away without using them and simplifying the circuit or it will affect the display in the long term?

Sorry for the dumb questions, I'm just curious.
Thanks again!

olikraus, 2018-10-08:
> I'm just curious.

https://en.wikipedia.org/wiki/Decoupling_capacitor

You can see it like this: A display is very complicated device. In fact it includes a very special kind of a microcontroller. As such, the display is a great source for noise on the supply lines. The caps avoid malfunction of other devices, connected to the same power supply (like your Arduino Board).

olikraus, 2018-10-08:
I once had the case, where the display did not work without the cap being placed next to the supply pins of the display. The noise / switching inside the display did cause so much noise, that the display itself did not start.

FrankX0, 2018-10-08:
The internal LCD supply (VLCD) is generated by a charge pump, relying on the capacitors between VB0+ and VB0- and between VB1+ and VB1-. You already found out these are quite important :-)

The 0.33uF capacitor is to buffer this generated power supply VLCD. Without it VLCD might become unstable, I a definitely would advise to keep this one in. The 10M resistor is described as "acting as a draining circuit when VDD is shut down abruptly". So it will discharge the 0.33uF capacitor when VDD is removed, most likely to protect the chip itself. Advise: keep it in.

The 10uF (bulk) and 0.1uF (HF) are used for decoupling. You might not (immediately) see any difference, but will prevent (intermittent) problems in the future.

mattogodoy, 2018-10-08:
Wow, that was fast!
Thank you guys for taking the time to answer. I've learned a lot :)

I'll close this issue as the problem has already been solved and hoping it might help someone else in a similar situation.

Keep it up!

closed 2018-10-08 by mattogodoy