olikraus/u8g2 #720: SSD0323 / OSRAM Pictiva support

deepfryed, 2018-10-05:
Hi I recently bought a few OSRAM Pictiva 128x64 OLEDs and was thinking about adding support for it in u8g2. Are there any development guides I should follow before I start hacking at it to send you a PR ?

I have some standalone code based on the documentation here [1] and it seems to work but I'd love to use it with u8g2.

Thanks

[1] http://www.ka7oei.com/programming_pictiva.html

deepfryed, 2018-10-08:
Thanks I'll give that a go!

olikraus, 2018-10-08:
I have created a new release with a new constructor for this, example `U8G2_SSD0323_OS128064_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

I also fixed the 8 pixel offset. Will it work better?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

deepfryed, 2018-10-09:
[image]

olikraus, 2018-10-09:
Looks good. Can you attach the modified c source also?

deepfryed, 2018-10-09:
[u8x8_d_ssd1325.c.txt](https://github.com/olikraus/u8g2/files/2458899/u8x8_d_ssd1325.c.txt)

olikraus, 2018-10-11:
Thanks... 
Code restructure...

I have created a new beta release 2.24.1, which should include your updates. Will it still work?

====

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2018-10-11:
:-)

closed 2018-10-11 by olikraus