olikraus/u8g2 #721: SSD1331 support?

Citrullin, 2018-10-05:
I am currently working on an application for a smartwatch. Since I am familiar with RIOT and this library is already implemented in RIOT I want to use this library. Looks really nice. But there is just one thing: It doesn't support the SSD1331 OLED. I guess it has the same interface as the SSD1306. So the only difference I see so far is the resolution and form factor. Here some code examples: https://github.com/curtpw/nRF5x-device-reverse-engineering/blob/a9d61eb83d640f13981463f4400f5a06d7eb1d73/X9-nrf52832-activity-tracker/Firmware/examples/nrf52_X9Project_ALPHABET/nrf52_X9Project_ALPHABET.ino

The datasheet can be found here: https://cdn-shop.adafruit.com/datasheets/SSD1331_1.2.pdf

Since I have other OLED modules laying around, I will try this lib with another module for now. Maybe I can implement it by myself and create a PR :)

olikraus, 2018-10-05:
SSD1331 displays are full color RGB displays. U8g2 is optimized for monochrome (black/white) displays. RGB support is not possible.

Citrullin, 2018-10-05:
Mhmm, okay. Therefore, I need to use something different. Okay, thanks.

closed 2018-10-05 by Citrullin

olikraus, 2018-10-06:
ssd1306 with yellow and blue is supported, if you mean those displays where the upper 8 rows are yellow and the other are blue (or opposite). These displays are still monochrome, but only the color of the OLED changes.