olikraus/u8g2 #722: Display pixels shifted to the left on SSD1306 SPI

flying4fun, 2018-10-08:
I'm having issues similar to issue #331.  I'm using the following constructor:
U8G2_SSD1305_128X32_NONAME_F_2ND_4W_HW_SPI u8g2(U8G2_R0, 26, 25, U8X8_PIN_NONE);

I did not have any issues using U8glib and this constructor:
U8GLIB_SSD1305_128X32 u8g(10, 6);

This is what I get with the following code:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_SSD1305_128X32_NONAME_F_2ND_4W_HW_SPI u8g2(U8G2_R0, 26, 25, U8X8_PIN_NONE);

void setup() {
  u8g2.begin();
  u8g2.setColorIndex(1);
  u8g2.drawBox(0,0,128,32);
  u8g2.setColorIndex(0);
  u8g2.drawBox(2,2,124,28);
  u8g2.setColorIndex(1);
  u8g2.drawBox(4,4,120,24);
  u8g2.sendBuffer();
}

void loop() { 
  ;
}
```

[image]

I saw on issue #331 a suggestion about trying SH1106 instead, but I don't see a 4 wire SPI 128x32 option for the SH1106.

flying4fun, 2018-10-08:
I vaguely recall having to modify the init string in U8glib.  I'll check my code from the old U8glib and try it in this new library.

flying4fun, 2018-10-09:
FIXED.

Don't know if my specific SSD1305 is different from others, but I had to modify the init_seq but still offset was wrong.  I changed the offset from 2 to 4:
/* default_x_offset = */ 4,
in the u8x8_d_ssd1305.c

And now it's working correctly.

closed 2018-10-09 by flying4fun

olikraus, 2018-10-09:
Ok :)