olikraus/u8g2 #724: SSD1305 controller listed, but the Adafruit 128x32 display dimension is not supported, please add

larsrengersen, 2018-10-13:
As mentioned in the FAQ I'm raising a request to add support for the Adafruit monochrome 2.3" 128x32 dimension and the SSD1305 controller.

Currently I'm using the U8X8_SSD1305_128X32_NONAME_4W_HW_SPI u8x8 line, but the first two columns of pixels are not being updated/addressed. Hope this is fixed by using the correct constructor.
[image]

olikraus, 2018-10-13:
Can you provide a link to the product?

larsrengersen, 2018-10-13:
https://www.adafruit.com/product/2675

olikraus, 2018-10-13:
ok...
https://github.com/adafruit/Adafruit_SSD1305_Library/blob/master/Adafruit_SSD1305.cpp#L189

larsrengersen, 2018-10-14:
Sorry, I do not understand the implications of that link.
On line 189 it says "#if defined SSD1306_128_32", but I have the SSD1305_128_32.
I'll try a different OLED than the one from Adafruit.

olikraus, 2018-10-16:
That was just for myself.

olikraus, 2018-10-16:
I have created beta release 2.24.2.
It includes the new constructor U8X8_SSD1305_128X32_ADAFRUIT_4W_HW_SPI 
Can you test this beta? Especially: Can you test the u8x8 flip mode example? Will it work correctly?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

larsrengersen, 2018-10-16:
Thanks!
I have just tested it and there are two things I notice:
1) While it flips I very shortly see random pixels
See https://imgur.com/gallery/fT3lDMi for a video
2) In Flipmode0 not the full F is visible and in Flipmode1 an extra black column (or two)
[image]
But I'm starting to think there is an issue with the screen itself since also with the Adafruit library I had two columns not being updated.

olikraus, 2018-10-16:
> But I'm starting to think there is an issue with the screen itself since also with the Adafruit library I had two columns not being updated.

no no, it is just a configuration problem.
Please locate file u8x8_d_ssd1305 of u8g2 library source code on your hard disk.

Line 250:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1305.c#L250

change the value from 0 to 2 or 4.
Which value will work?

larsrengersen, 2018-10-16:
Thanks! It works!
`  /* default_x_offset = */ 4,`
did the trick.
[image]
Also tested the "GraphicsTest" example and on the screen with all numbers and letters, all characters are visible.

olikraus, 2018-10-28:
thanks for the feedback... fixed...

closed 2018-10-28 by olikraus