olikraus/u8g2 #726: U8g2 lib freezes ESP32 after some time

Isopodus, 2018-10-15:
_Dear olikraus,
thank you for this great library for various types of LCDs._

I've got some problems while using it on my **DOIT ESP32 DEVKIT v1** and **ST7920 128x64 LCD**. 
**Here is my testing code:**

```
#include <U8g2lib.h>
#include "SD.h"
#include "FS.h"
U8G2_ST7920_128X64_1_SW_SPI lcd(U8G2_R0, 14, 13, 27, 26);

void setup() {
  //Serial.end();
  SD.begin();
  lcd.begin();
}

unsigned int mil = 0;
int counter = 0;

void loop() {
  if(millis() - mil > 5000) //writing seconds to file every 5 sec
  {
    File file = SD.open("/file.txt", FILE_APPEND);
    file.println(String(String(counter) + "\t" + String(millis() / 1000)));
    file.close();
    mil = millis();
    counter ++;
  }
  lcd.firstPage(); //displaying seconds on screen
  do {
    lcd.setFont(u8g2_font_profont10_tr);  
    lcd.drawStr(10, 10, String(millis() / 1000).c_str());  
  } while (lcd.nextPage());
}
```

The problem is, that after some time of this code running (**1:11:30** or 2:23:00 or 3:34:30 and so on (multiple of **1:11:30**)), the ESP32 freezes for some time (exactly same **1:11:30** and multiple of it).

Seems like void loop() gets stuck somewhere; in file log the next line is more than 5 seconds after previous,  seconds on screen are frozen too. When I **cut** all the LCD stuff and leave only file log, everything works fine, there aren't any hangs in log, so it maybe there are some problems with u8g2 library. Do you know anything about this issue? Thanks.

_Eugene_

olikraus, 2018-10-15:
hmm... There was a HW SPI problem (see here: #445) but you use SW SPI, so this does not appy.
Maybe some garbage collection happens, due to the String() constructor. Actually I have never used String()... probably because I do not really trust these C++ classes.
Instead of
```
String(millis() / 1000).c_str()
```
you could use
```
char buf[32];
sprintf(buf, "%d", millis() / 1000);
```

Isopodus, 2018-10-15:
The problem is, that whole my project is running on C++ String class. The String c_str() function just returns char *, so this can't really affect so much. I've never seen any garbage problems while usnig this class...

olikraus, 2018-10-15:
>  I've never seen any garbage problems while usnig this class...

Is this a prove, that these problems do not exist?

String allocates memory on the heap. This is a fact.

Isopodus, 2018-10-16:
Okay, thank you for the answer, maybe i should check if it works properly with C-type strings...

olikraus, 2018-10-28:
feel free to reopen this for further questions/discussion.

closed 2018-10-28 by olikraus