olikraus/u8g2 #727: Russian text and my font)

neoblack2, 2018-10-24:
I have sketch. 
I have a sketch that I made one year ago and which displayed Russian text. That opened it now and updated u8g2. Russian text is no longer displayed. What to do?

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 53, /* reset=*/ 49);

const uint8_t haxcorp[1501] U8G2_FONT_SECTION("haxcorp") PROGMEM = { 
  "\204\0\3\2\4\4\2\4\5\10\13\0\376\11\0\10\377\1=\0\0\2\260 \5\0\242\4!\7q\242"
  "\304\240\4\42\7\63.Eb\11#\13U\242MeP*\203R\1$\16\225\236U\266T\224lK\224"
  "\312\26\1%\14w\42N\250D\255\245\26\61\1&\15v\342\215\26\265IR\242EK\0'\6\61\256"
  "\304\0(\10\223\36UR\352V)\11\223\36EV\352R\2*\12U\246U\245\262\64E\0+\12U"
  "\242U\30\15R\30\1,\7\62\336L\242\0-\6\25\252\305 .\6\21\242D\0/\10w\42v\332"
  "g\0\60\14u\242\315\222IK\62i\311\2\61\7r\342\314\322\3\62\12u\242\315\222\205Y\333 \63"
  "\14u\242\315\222\205\221\252%\13\0\64\14u\242]&%\245d\320\302\4\65\13u\242\305qHC-"
  "Y\0\66\14u\242\315\222\211C\222i\311\2\67\13u\242\305 \326\302,\214\0\70\14u\242\315\222i"
  "\311\222i\311\2\71\14u\242\315\222i\311\20j\311\2:\6A\242D\24;\10b\336L\232(\0<"
  "\7S\42UR+=\10\65\246\305\240\16\2>\10S\42EV)\1?\13u\242\315\222\205Y\35\212"
  "\0@\23\210\42\316\260\304\322\242D\211\22%\322\220\344\310\260\0A\11D\267\215\22I\211\2B\14t"
  "\243E&-\303\220H\31\0C\13t\243]\244\14C\62i\1D\11U\246U\230M\313 E\12U"
  "\242\305\240L[\30\1F\14f\242\225\26%\211K)\223\0G\14f\242\225\26%\241\230D\231\4H"
  "\12U\246U\226\15J\32\1I\12u\242U\266\64\205M\0J\11U\246U\232\14Z\11K\12u\242"
  "U\330\224T\266\10L\11\65\252MeP*\0M\13u\242U\266\64U*[\4N\13\205\242U\226"
  "\15JT\11\33O\14\205\242U\232\14SR\12\213\0P\13\205\242eSR\32\226\64\2Q\13u\242"
  "EX\211*\203V\2R\12U\246U\66\250C\24\1S\16u\242U\226$\226R\22%Q\222\0T"
  "\12U\246U\64\244\203\26\1U\15u\242M)\211\222(I,\265\10V\16\205\242U\226$\226R\222"
  "Xj\21\0W\7D\247\305\17\1[\10\223\36\305\20\365i\134\17w\42F\16\344@\16\344@\16\344"
  "@\0]\10\223\36\305\324\247!^\10\65\256U\226\324\2_\6\25\242\305 `\6\63.EV{\12"
  "\223\36U\22U\262\250\26|\6\221\236\304\3}\13\223\36E\26\325\222\250\22\1~\7&\352\215d\1"
  "\251\17w\42\316\240\244\312\242d\312\242&\203\2\261\14u\242U\30\15R\230\3\203\0\367\12U\242U"
  "\16\14:\20\1\0\0\4\20\14u\242\315\222i\303\220\331\2\4\21\15u\242\305\61\34\222L\33\24\0"
  "\4\22\16u\242\305\220d\332\240d\332\240\0\4\23\11u\242\305\261G\0\4\24\20\207\36\326V\312\242"
  ",\312\242,\31\6\65\4\25\14u\242\305\61\34\222\60\34\4\4\26\21w\42F\24I\221T\31\224("
  "\222\42\251\0\4\27\15u\242\315\222\205\311\252%\13\0\4\30\14u\242EfZ\222I\323\2\4\31\15"
  "\205\246\315\222\231\226d\322\264\0\4\32\16u\242E&%%-\211*Y\0\4\33\17u\242\325\22%"
  "Q\22%Q\22i\1\4\34\15w\42F\272-\25)R\255\1\4\35\13u\242Ef\33\206\314\26\4"
  "\36\13u\242\315\222yK\26\0\4\37\11u\242\305\315\267\0\4 \15u\242\305\220d\332\240\204E\0"
  "\4!\13u\242\315\222\211m\311\2\4\42\12u\242\305 \205=\1\4#\15u\242EfK\206PK"
  "\26\0\4$\23\227\36^\70(Q$ER$ERe\20\63\0\4%\14u\242E\246%\265JM"
  "\13\4&\21\206\336E\226dI\226dI\226d\311\260\6\4'\13u\242EfK\206\260\1\4(\21"
  "w\42F\24I\221\24I\221\24I\321\60\10\4)\23\210\36FT\211*Q%\252D\225\250\62\14:"
  "\20\4*\16w\42\306\134\36\262R\26\15\11\0\4+\16w\42Fj\35$-\321\222A\12\4,\15"
  "u\242EX\34\222L\33\24\0\4-\15u\242\315\222\205\321\250%\13\0\4.\21x\42F\266D\231"
  "\224\15\232\224I\231\266\0\4/\15u\242\315\240i\311\20%%-\4\60\12Tb\315\20\231\222\1\4"
  "\61\14tb\315\220-\221)Q\0\4\62\14Tb\305\22\15I\64$\0\4\63\11Tb\305\240\265\1"
  "\4\64\14f\336\225\26\365\62\14a\0\4\65\12Tb\215\22\15\243\2\4\66\16W\42F\24I\225A"
  "\211\42\251\0\4\67\13Tb\215\22U\242D\1\4\70\13U\242E&-\311\244\5\4\71\15\205\242M"
  "-\7\62iI&-\4:\13TbE\244$RR\12\4;\11Tb\225\322\213\24\4<\13U\242"
  "E\266,\211\246\5\4=\12TbE$\15\223\24\4>\12Tb\215\22\231\22\5\4?\11Tb\305"
  " \71\5\4@\14tZ\305\22\231\206$\313\0\4A\13Tb\215\22iQ\242\0\4B\12U\242\305"
  " \205M\0\4C\13tZE\344\224l\211\2\4D\21\227\32^\34\16J\24I\221T\31\304\70\3"
  "\4E\13TbE$%J$\5\4F\16e\236E\224DI\224D\311 \6\4G\12TbE$"
  "%[\1\4H\16W\42F\24I\221\24I\321\60\10\4I\20h\36FT\211*Q%\252\14\203\16"
  "\4\4J\12U\242\205\30N\225\5\4K\15V\342E(\16\221\224\14Q\0\4L\13TbE\226-"
  "\321\220\0\4N\15V\342E\244\224\206H\211,\0\4O\13Tb\315\20%K\42\5\0";

void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();		// enable UTF8 support for the Arduino print() function
}

void loop(void) {
  u8g2.setFont(haxcorp);  
  u8g2.clearBuffer();
  u8g2.setCursor(0, 40);
  u8g2.print("привет");		
  u8g2.sendBuffer();
  delay(1000);
}
```

neoblack2, 2018-10-24:
Now worked! Thx

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 53, /* reset=*/ 49);

const uint8_t haxcorp[1505] U8G2_FONT_SECTION("haxcorp") = 
    "\204\0\3\2\4\4\2\4\5\10\13\0\376\11\0\10\377\1=\0\0\2\260 \5\0\242\4!\7q\242"
  "\304\240\4\42\7\63.Eb\11#\13U\242MeP*\203R\1$\16\225\236U\266T\224lK\224"
  "\312\26\1%\14w\42N\250D\255\245\26\61\1&\15v\342\215\26\265IR\242EK\0'\6\61\256"
  "\304\0(\10\223\36UR\352V)\11\223\36EV\352R\2*\12U\246U\245\262\64E\0+\12U"
  "\242U\30\15R\30\1,\7\62\336L\242\0-\6\25\252\305 .\6\21\242D\0/\10w\42v\332"
  "g\0\60\14u\242\315\222IK\62i\311\2\61\7r\342\314\322\3\62\12u\242\315\222\205Y\333 \63"
  "\14u\242\315\222\205\221\252%\13\0\64\14u\242]&%\245d\320\302\4\65\13u\242\305qHC-"
  "Y\0\66\14u\242\315\222\211C\222i\311\2\67\13u\242\305 \326\302,\214\0\70\14u\242\315\222i"
  "\311\222i\311\2\71\14u\242\315\222i\311\20j\311\2:\6A\242D\24;\10b\336L\232(\0<"
  "\7S\42UR+=\10\65\246\305\240\16\2>\10S\42EV)\1\77\13u\242\315\222\205Y\35\212"
  "\0@\23\210\42\316\260\304\322\242D\211\22%\322\220\344\310\260\0A\11D\267\215\22I\211\2B\14t"
  "\243E&-\303\220H\31\0C\13t\243]\244\14C\62i\1D\11U\246U\230M\313 E\12U"
  "\242\305\240L[\30\1F\14f\242\225\26%\211K)\223\0G\14f\242\225\26%\241\230D\231\4H"
  "\12U\246U\226\15J\32\1I\12u\242U\266\64\205M\0J\11U\246U\232\14Z\11K\12u\242"
  "U\330\224T\266\10L\11\65\252MeP*\0M\13u\242U\266\64U*[\4N\13\205\242U\226"
  "\15JT\11\33O\14\205\242U\232\14SR\12\213\0P\13\205\242eSR\32\226\64\2Q\13u\242"
  "EX\211*\203V\2R\12U\246U\66\250C\24\1S\16u\242U\226$\226R\22%Q\222\0T"
  "\12U\246U\64\244\203\26\1U\15u\242M)\211\222(I,\265\10V\16\205\242U\226$\226R\222"
  "Xj\21\0W\7D\247\305\17\1[\10\223\36\305\20\365i\134\17w\42F\16\344@\16\344@\16\344"
  "@\0]\10\223\36\305\324\247!^\10\65\256U\226\324\2_\6\25\242\305 `\6\63.EV{\12"
  "\223\36U\22U\262\250\26|\6\221\236\304\3}\13\223\36E\26\325\222\250\22\1~\7&\352\215d\1"
  "\251\17w\42\316\240\244\312\242d\312\242&\203\2\261\14u\242U\30\15R\230\3\203\0\367\12U\242U"
  "\16\14:\20\1\0\0\0\4\377\377\4\20\14u\242\315\222i\303\220\331\2\4\21\15u\242\305\61\34\222"
  "L\33\24\0\4\22\16u\242\305\220d\332\240d\332\240\0\4\23\11u\242\305\261G\0\4\24\20\207\36"
  "\326V\312\242,\312\242,\31\6\65\4\25\14u\242\305\61\34\222\60\34\4\4\26\21w\42F\24I\221"
  "T\31\224(\222\42\251\0\4\27\15u\242\315\222\205\311\252%\13\0\4\30\14u\242EfZ\222I\323"
  "\2\4\31\15\205\246\315\222\231\226d\322\264\0\4\32\16u\242E&%%-\211*Y\0\4\33\17u"
  "\242\325\22%Q\22%Q\22i\1\4\34\15w\42F\272-\25)R\255\1\4\35\13u\242Ef\33"
  "\206\314\26\4\36\13u\242\315\222yK\26\0\4\37\11u\242\305\315\267\0\4 \15u\242\305\220d\332"
  "\240\204E\0\4!\13u\242\315\222\211m\311\2\4\42\12u\242\305 \205=\1\4#\15u\242Ef"
  "K\206PK\26\0\4$\23\227\36^\70(Q$ER$ERe\20\63\0\4%\14u\242E\246"
  "%\265JM\13\4&\21\206\336E\226dI\226dI\226d\311\260\6\4'\13u\242EfK\206\260"
  "\1\4(\21w\42F\24I\221\24I\221\24I\321\60\10\4)\23\210\36FT\211*Q%\252D\225"
  "\250\62\14:\20\4*\16w\42\306\134\36\262R\26\15\11\0\4+\16w\42Fj\35$-\321\222A"
  "\12\4,\15u\242EX\34\222L\33\24\0\4-\15u\242\315\222\205\321\250%\13\0\4.\21x\42"
  "F\266D\231\224\15\232\224I\231\266\0\4/\15u\242\315\240i\311\20%%-\4\60\12Tb\315\20"
  "\231\222\1\4\61\14tb\315\220-\221)Q\0\4\62\14Tb\305\22\15I\64$\0\4\63\11Tb"
  "\305\240\265\1\4\64\14f\336\225\26\365\62\14a\0\4\65\12Tb\215\22\15\243\2\4\66\16W\42F"
  "\24I\225A\211\42\251\0\4\67\13Tb\215\22U\242D\1\4\70\13U\242E&-\311\244\5\4\71"
  "\15\205\242M-\7\62iI&-\4:\13TbE\244$RR\12\4;\11Tb\225\322\213\24\4"
  "<\13U\242E\266,\211\246\5\4=\12TbE$\15\223\24\4>\12Tb\215\22\231\22\5\4\77"
  "\11Tb\305 \71\5\4@\14tZ\305\22\231\206$\313\0\4A\13Tb\215\22iQ\242\0\4B"
  "\12U\242\305 \205M\0\4C\13tZE\344\224l\211\2\4D\21\227\32^\34\16J\24I\221T"
  "\31\304\70\3\4E\13TbE$%J$\5\4F\16e\236E\224DI\224D\311 \6\4G\12"
  "TbE$%[\1\4H\16W\42F\24I\221\24I\321\60\10\4I\20h\36FT\211*Q%"
  "\252\14\203\16\4\4J\12U\242\205\30N\225\5\4K\15V\342E(\16\221\224\14Q\0\4L\13T"
  "bE\226-\321\220\0\4N\15V\342E\244\224\206H\211,\0\4O\13Tb\315\20%K\42\5\0"
  "";

void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();		// enable UTF8 support for the Arduino print() function
}

void loop(void) {
  u8g2.setFont(haxcorp);  
  u8g2.clearBuffer();
  u8g2.setCursor(0, 40);
  u8g2.print("привет");		
  u8g2.sendBuffer();
  delay(1000);
}
```

olikraus, 2018-10-24:
Great. Also note that the haxcorp font is also part of u8g2 now:
https://github.com/olikraus/u8g2/wiki/fntgrpfontstruct#haxrcorp-4089