olikraus/u8g2 #729: Converting from bitmap array to BDF

Maldus512, 2018-10-28:
Hello,
I'm using your library to drive a display with an ESP32. I need to show strings in cyrillic, but none of the available fonts suit my needs (8x8 monospace font).
The board I'm programming is a newer version of a previous product that already had russian strings displayed. It didn't use a proper graphic library, so the font is a simple bitmap array.

I've tried generating a .bdf file from said array with a custom script and then to convert again to c with your utility (bdfconv), but I receive the following error:

    Glyph size problem: encoding=12/0xc, width=8, height=8, requested position x=8, y=4 (use BBX 9 ...?)
    bdfconv: bdf_glyph.c:177: bg_SetBitmapPixel: Assertion `x < ((bg->bitmap_width+7)/8)*8' failed.
    Aborted (core dumped)

I'm not sure I understand what the problem is, maybe you can help me?

This is a use case I might face again in the future since I work for an employer that has a lot of those old low-level projects with fonts as bitmap arrays.
I must admit I'm not familiar with BDF or font editing in general. I was wondering if I was doing a very obvious mistake I'm oblivious to. Otherwise I'm probably better off creating the .bdf through an actual editor based on the bitmap array.

This is the bitmap array: [font.txt](https://github.com/olikraus/u8g2/files/2522876/font.txt)
This is the BDF I generated: [test.txt](https://github.com/olikraus/u8g2/files/2522878/test.txt)

Thank you for your time (and for the great library).

olikraus, 2018-10-28:
The problem is located in the bitmap of encoding 12: the pattern ffffe0 is too big to fit into 8 pixel

Maldus512, 2018-10-28:
Probably a mistake in my conversion script. Many thanks, correcting them lead to a properly generated C font

olikraus, 2018-10-29:
:-)

closed 2018-12-20 by olikraus