olikraus/u8g2 #732: Do I still need GxEPD2 ?

SunboX, 2018-10-29:
Hi,

I came across your great library when using https://github.com/ZinggJM/GxEPD2

A question came across my mind. I only need to draw lines, dots and text on a 7.5" Waveshare e-Paper Display. 

Do I still need GxEPD2 for that or is your library enough? 

Greetings and thanks for your great work! 
André

olikraus, 2018-10-30:
Yes. Although U8g2 support a (very) small number of e-paper devices, the current support is limited and I do not really work on further improvement of e-paper devices. Main reason is the U8g2 internal software architecture, which is not optimized for e-paper devices. 

If you need access to the U8g2 fonts, you can use https://github.com/olikraus/U8g2_for_Adafruit_GFX along with https://github.com/ZinggJM/GxEPD2.

SunboX, 2018-10-30:
Hey, thanks for the info! So I will stick with GxEPD2. 🙂

closed 2018-10-30 by SunboX