olikraus/u8g2 #733: Why will overlap, not found in the specified location screen clearing function

xfdr0805, 2018-10-30:
**u8g2.setCursor(0, row_2);**
    **u8g2.print("IP:conntcting...");**
    u8g2.sendBuffer();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    Serial.print("\nWiFi is Connecting");
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(100);
        //Serial.print(".");
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    }
    Serial.println(WiFi.localIP());
    **u8g2.setCursor(7 * 3, row_2);
    u8g2.print(WiFi.localIP());**
    u8g2.sendBuffer();

olikraus, 2018-10-30:
https://github.com/olikraus/u8g2/wiki/u8g2reference#clearbuffer

xfdr0805, 2018-10-30:
> https://github.com/olikraus/u8g2/wiki/u8g2reference#clearbuffer

I know this, but this function will remove the entire screen all, I want to just clear the contents of the specified location，thank you

olikraus, 2018-10-30:
You can [draw](https://github.com/olikraus/u8g2/wiki/u8g2reference#drawbox) a box with [color 0](https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor).

xfdr0805, 2018-10-30:
> You can [draw](https://github.com/olikraus/u8g2/wiki/u8g2reference#drawbox) a box with [color 0](https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor).

Thank you very much, this should be feasible, but I used the space to cover the content at the time, the actual can not be cleared, whether to give this function a separate name, the first thing that comes to mind under normal circumstances is to clear the specified location content (ex clearbufferAt), Instead of DrawBox