olikraus/u8g2 #734: sendBuffer() timeout

ferchinas, 2018-10-31:
Hello:
CPU: ESP32
Display: Oled wemos SSD1306 64x48 I2C (0.66)
Refresh: 4 fps (250ms)
Cons: U8G2_SSD1306_64X48_ER_F_HW_I2C
SDA: Gpio21
SCL: Gpio22

I'm using the library and it was working wonders, sendBuffer () normally takes between 14 and 18ms.
But suddenly the sendBuffer () method takes 1800ms. Once it does, it continues to delay 1800ms on each call to sendBuffer () until I perform a general reset.
Perform the display update but during that time block the execution of the program.

I have tested with 4 Wemos Oled and with an SSD1306 (0.96) with the same results
Could you help me find the error?
Thank you

olikraus, 2018-11-01:
I think it is a ESP32 issue. Did you change ESP32 version?

ferchinas, 2018-11-01:
That's right, it's an ESP32 issue.
Thank you very much for the response and congratulations for the library.

closed 2018-11-01 by ferchinas

ferchinas, 2018-11-05:
Hi olikraus:
I'm trying to patch this situation.
I have noticed that restarting the microcontroller (without interrupting the display power) the problem is solved.
When I detect that it takes too long to process the sendBuffer () method, I would like to restart the display.
But calling the begin () method again blocks the display.
Is there a method to restart the display or i2c communication?
Thank you