olikraus/u8g2 #736: Only update the changed pixels in full buffer mode

cornedoggen, 2018-11-01:
Hi all, 

I have a question about the update procedure of the u8g2 library. 
When you use the full buffer mode (f extension in constructor), can it be configured to only update the pixels/bytes which are changed since last time? For example: 
I have some text down in the display and an audio level meter in the top. The level meter should be updated very fast to give a smooth response for the user. If you need to redraw the entire display for every audio metering update, then the meter will behave too slow... :(. Therefore i think it is possible to only update the region / area which is changed. In other library which i used in the past ( RAMTEX - https://www.ramtex.dk  ), they call this the 'dirty area'. It is very very fast! because if you change only a specific item/widget on the screen it updates that area!. The dirty area is a rectangle with (x0,y0, x1,y1) which automatically scales when you call the library functions for writing text, or drawing lines. When you call the send-buffer() function it should send the data bytes which are surrounded by the dirty area. What do you think about this feature? 

Kind Regards, 

Corné Doggen

olikraus, 2018-11-02:
Thanks for comparing with RAMTEX library. 

With respect to the dirty window: Such a feature would be thinkable with some restrictions. It can be implemented for full buffer mode only and it will increase flash ROM size. 

I the size and position of this window is already known (should be true in your case), then this should be possible right now. hmmm in fact this should have no RAM/ROM impact for users, who do not use this function...

The use case would be this: You update a part of the screen say x0, y0, x1, y1.
Then there would be a function, which sends only this part to the display:
U8G2:sendWindow(x0, y0, x1, y1)
Would this be a solution? I mean, there would be no automatic scaling of the dirty window, so comfort is lesser, but I think this is not required in your case (because you know your dirty window in advance).

Another impact is this: Such an update is only possible for displays with support the u8x8 API (ok, should be true for most of the displays). But to ensure that this works for your: What is your display type?
One more restriction is the fact, that the coordinates of the dirty window, must be dividable by 8.
This means, the actual window has to be U8G2:sendTileWindow(x0/8, y0/8, (x1+7)/8, (y1+7)/8) due to the memory layout of these black/white displays.

olikraus, 2018-11-02:
I have implemented the above mentioned sub block transfer function here:
https://github.com/olikraus/u8g2/blob/master/sys/sdl/full_buffer_send_window/main.c#L15
and here:
https://github.com/olikraus/u8g2/blob/master/sys/sdl/full_buffer_send_window/main.c#L42

Here is the code which can be copied into your project.
```
/*
  Limitations:
    - Tile positions and sizes (pixel position divided by 8)
    - Any display rotation/mirror is ignored
    - Only works with displays, which support U8x8 API
    - Only available in full buffer mode
*/
u8g2_SendTileSubBuffer(u8g2_t *u8g2, uint8_t x, uint8_t y, uint8_t w, uint8_t h)
{
  uint16_t page_size;
  uint8_t *ptr;

  /* the page size in bytes is equal to the pixel width in bits */
  page_size = u8g2->pixel_buf_width;
    
  ptr = u8g2_GetBufferPtr(u8g2);
  ptr += x*8;
  ptr += page_size*y;
  
  while( h > 0 )
  {
    u8x8_DrawTile( u8g2_GetU8x8(u8g2), x, y, w, ptr );
    ptr += page_size;
    y++;
    h--;
  }  
}

/*
  sub buffer window in pixel coordinates
  lower left corner is NOT included, this means the transfer range is from
  x0..x1-1
  y0..y1-1
*/
void u8g2_SendWindow(u8g2_t *u8g2, u8g2_uint_t x0, u8g2_uint_t y0, u8g2_uint_t x1, u8g2_uint_t y1)
{
  x0 /= 8;
  y0 /= 8;
  x1 += 7;
  x1 /= 8;
  y1 +=7;
  y1 /= 8;
  u8g2_SendTileSubBuffer(u8g2, x0, y0, x1-x0, y1-y0);
}
```

For calling these functions within Arduino/C++ environment: Get the pointer to the u8g2 structure with U8G2:getU8g2(), so it will look like this:
```
u8g2_SendWindow(u8g2.getU8g2(), x0, y0, x1, y1); 
```

olikraus, 2018-11-02:
ToDo: Move example code to production code.

cornedoggen, 2018-11-05:
Hi Oli,

Thank you so much for this update Oli! great!

In my case i have to update only the area on top of the display (128x128 SSD1327). 
There are the audio level meters located, and i know the coordinates of that area. 
So , the missing autoscale function is not a problem for me at the moment. 

As a software developer i try to fully understand your implementation of the library. 
I have a few questions about it: 

- What is the representation of:  Page, Tile, Row ?? and how are they related? 
I would love to see a picture of display and indicate what is a page, tile, and row. 
I already understand this is dependent from the constructor method... 
The constructor i am using is the following: 
 - u8g2_Setup_ssd1327_midas_128x128_f()

In the constructor it allocates memory with: 
- u8g2_m_16_16_f().

What are the 16, 16 telling me?? the 'f' is full buffer, thats clear to me. 
Is it 16 by 16 pixels/bytes? is 16x16 the Tile size? Are Tiles in horizontal or vertical direction?
So, main question is.. how do the terms Page, Tile, and Row look like in a graphical representation in my mind?

Kind Regards, 
Corné

olikraus, 2018-11-05:
> What is the representation of: Page

This comes from the display controller datasheets. The memory of most black/white displays is organized as a list of pages. So for a full understanding, it is probably required to read the datasheets, like the SSD1306 datasheet.

But simply spoken, a page is a memory area on the display, which has a height of 8 pixel and a width which is identical to the width of the display.
So for example a 128x64 pixel display, usually has a memory, which is organized into 8 pages of 
8x128 pixel each.

> Tile

A tile a quadratic area on the display, which contains 8x8 = 64 pixel. A page is a sequence of tiles, organized from left to right. So, the above page with 8x128 pixel is composed of 16 tiles.

> I would love to see a picture of display and indicate what is a page, tile, and row.

There is a brief description in the reference manual.
The table depicts the left part of the first (topmost) page of the display:
https://github.com/olikraus/u8g2/wiki/u8g2reference#memory-structure-for-controller-with-u8x8-support

> Row

Well, not so clearly defined, but usually a tile or buffer row.
A buffer contains the memory for one page (1-constructor), two pages (2-constructor) or all pages (f-constructor)

So for a 128x64 display and the 1-constuctor, there are 8 rows (0..7). For the same display with the 2-constructor, there will be only 4 rows (0..3).

For the f-constructor (like in your case), the whole page and row topic does not make much sense, because the buffer will contain the complete frame buffer for the display.

> u8g2_m_16_16_f()

U8g2 does not use malloc(), so it uses several different memory allocation procedures. U8g2 depends on the linker garbage collector: It is assumed, that functions, which are not called, are removed during linking stage.

The name includes the size of the memory in tiles. So in this case, a memory with 16x16 tiles are allocated. These are 16*8 x 16*8 = 128x128 pixel = 128x128/8 Bytes = 2048 Bytes: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_d_memory.c#L126

> Are Tiles in horizontal or vertical direction?

Yes, tiles are horizontals blocks of 8x8 pixel each

> how do the terms Page, Tile, and Row look like in a graphical representation in my mind?

The sdl u8g2 emulation will show the tiles as checkerboard (maybe little bit difficult to see).
The topmost row of this checkerboard is one page.

[image]

hope this helps...

cornedoggen, 2018-11-05:
Excellent explanation!! 

I did a git pull to update my u8g2 sources, but i can't find the new functions you added:
- u8g2_SendWindow()
- u8g2_SendTileSubBuffer()

Or is it intentionally leaved out of the repo ? and i need to copy the code manually to my project?

In total i have connected 8 displays to my microcontroller via a SPI buss with 8 separate CS lines. So to initialize all the displays i create 8 instances of u8g2_t in an array:

`struct displayInfo[8]
`u8g2_t display[8]`

To initialize the displays i call the corresponding constructor with the 'f' for all the displays. for every function call of the constructor i give the same callback references as the parameters. So all the calls from your library end in 2 callback functions, the gpio_and_delay_cb, and byte_cb. 
And now the problem arises: In the callback function i have no idea about which displays i have to control. For example.. i need to control the CS line of the display, but how do i know which one??

In most libraries this issue is solved by an extra void *userdata parameter in the callback function. 
This pointer is also available in the constructor and is referenced to an instance / object in your code. 

`void u8g2_Setup_ssd1327_midas_128x128_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, void *byte_cb_userdata, u8x8_msg_cb gpio_and_delay_cb, void *gpio_and_delay_cb_userdata)
uint8_t *gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr, void *user_data);
uint8_t *byte_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr, void *user_data);`

In my case it would be a reference to (void*)&displayInfo[0] for the first display. In the displayInfo struct the CS line pin can be found. which result to something like this in the callback:

`uint8_t *gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr, void *user_data)
{
        struct displayInfo *info = (struct displayInfo *)user_data;

	switch(msg)
	{
		case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
			/* Select all the 6 displays simultaneously */ 
			ioport_set_pin_level( info->cs_pin, arg_int );
			break;
	}
	return 1;
}`

Kind Regards, 

Corné

olikraus, 2018-11-05:
> Or is it intentionally leaved out of the repo ? and i need to copy the code manually to my project?

The functions are planed, but not part of the csrc directory. I added the functions to the sdl emulation directory:
https://github.com/olikraus/u8g2/blob/master/sys/sdl/full_buffer_send_window/main.c#L15

So the code is in "sys/sdl/full_buffer_send_window" as part of the repo already.

> And now the problem arises: In the callback function i have no idea about which displays i have to control. 

You need to create 8 different gpio&delay procedures. Actually these gpio&delay procedures are custom user procedures, which need to be adjusted to your needs. In fact I do not know which and from where you took these procedures, but in general, you need to catch the msg (given as agument) and react accordingly. Here is an example for a gpio&delay procedure for display 3:

```
uint8_t *gpio_and_delay_cb_disp_3(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
switch(msg)
{
	case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
		ioport_set_pin_level( disp-3-cs-pin, arg_int );
		break;
	default:				// CS (chip select) pin: Output level in arg_int
		return gpio_and_delay_cb(u8x8, msg, arg_int, arg_ptr);
}
return 1;
}
```
So, instead of passing `gpio_and_delay_cb` to the constructor, use `gpio_and_delay_cb_disp_3`.

Details are documented here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

Ok, the above mentioned option allows you to customize u8g2 for all 8 displays without additional user pointer. U8x8 (and u8g2) also support a custom user pointer. However, you need to enable this in u8x8.h: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L95

Once defined, you can use set and get the user pointer:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L358

You can set the user pointer to a variable (uint16_t in the example below) with the pin number, and get the number in the gpio&delay procedure. I think it would look like this:

```
uint8_t *my_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
switch(msg)
{
	case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
		ioport_set_pin_level( *((uint16_t *)u8x8_GetUserPtr(u8x8)), arg_int );
		break;
	default:				// CS (chip select) pin: Output level in arg_int
		return gpio_and_delay_cb(u8x8, msg, arg_int, arg_ptr);
}
return 1;
}
```
The advantage will be, that you need to create only one gpio and delay procedure for all displays.

Then there is a third method: U8x8 can be compiled with an array, which can be used to store pin numbers. This is required for the Arduino environment, but could be used also by you.
You need to compile with -DU8X8_USE_PINS:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L352

Once the array is added, you can assign the value of the cs pin to
u8x8->pins[U8X8_PIN_CS]. Then the gpio&delay procedure will look like this:

```
uint8_t *my_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
switch(msg)
{
	case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
		ioport_set_pin_level( u8x8->pins[U8X8_PIN_CS], arg_int );
		break;
	default:				// CS (chip select) pin: Output level in arg_int
		return gpio_and_delay_cb(u8x8, msg, arg_int, arg_ptr);
}
return 1;
}
```
> In my case it would be a reference to (void*)&displayInfo[0] for the first display. In the displayInfo struct the CS line pin can be found. which result to something like this in the callback:

If you already plan to use a displayInfo struct, then the use of the USER_PTR might be the best approach.

1. Add -DU8X8_WITH_USER_PTR to your compiler call.
2. Create a custom gpio&delay procedure:

```
uint8_t *my_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
struct displayInfo *info = (struct displayInfo *)u8x8_GetUserPtr(u8x8);

switch(msg)
{
	case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
		ioport_set_pin_level( info->cs_pin, arg_int );
		break;
	default:				// CS (chip select) pin: Output level in arg_int
		return gpio_and_delay_cb(u8x8, msg, arg_int, arg_ptr);
}
return 1;
}
```
3. During init phase, assign the pointer to displayinfo struct via u8x8_SetUserPtr() 

```
struct displayInfo[8]
u8g2_t display[8]
for( i = 0; i < 8; i++ )
{
	u8g2_Setup_ssd1327_midas_128x128_f( &(display[i]), U8G2_R0, byte_cb, my_gpio_and_delay_cb);
	u8x8_SetUserPtr(u8g2_GetU8x8(&(display[i])), (void *)&(display[i]) );
}
```

cornedoggen, 2018-11-05:
Hi Oli, 

Thank you so much again for the fast and comprehensive reply! Really appreciate that! :)

The hole idea behind asking for a user pointer was offcourse avoiding to have 8x times the same callback. I did not mention that, im sorry, but you understood already. i didn't knew the u8x8 object was designed to hold a reference user pointer. I will use that to point to my info struct! 

Thumbs up for u8g2! Great and well designed library!

Corné

olikraus, 2018-11-05:
> I will do it with malloc

Sure, just update u8g2_m_16_16_f(): Call malloc and return a pointer to the allocated memory.

> to store all 8 display buffers into internal RAM

With the current default behavior, the memory would be shared among all 8 displays.

cornedoggen, 2018-11-05:
to store all 8 display buffers into internal RAM

With the current default behavior, the memory would be shared among all 8 displays

? This last sentence i don't understand, since if you call the function u8g2_m_16_16_f() it allocates everytime you call it new memory.. right? and it should, because every display has other content. can you explain what you mean with 'sharing the memory among 8 displays' ?

CDoggen, 2018-11-05:
There is no need to store the content of each display in the memory of the controller

Offcourse this IS needed, Since the content of Every display is different. If all the displays would use the same memory, then writing to a display requires a full drawn of the display. This is not what i want. I want to update only the Area in the buffer what is charged en witte Those changes to the display. 

I dont understand why You say buffering is nit needed.... since the 3 constructor methods do use buffering! What do You mean with this?

Regarding the buf pointer in u8x8, i ment the u8g2. You are right.

e-music, 2018-11-08:
I have added the functions, but not sure how the calculation is done to convert coordinates from pixels to tiles. For example, suppose I want to draw a symbol at a certain coordinate like this:
`u8g2_DrawStr(&u8g2, 220, 40, "\xD9");`

How to calculate the coordinates (in tiles) to update the area used by that symbol then pass that to the **u8g2_SendWindow()** function?

olikraus, 2018-11-08:
Precondition: R0 is used in the constructor

To get the tile coordinates from the pixel position, divide x and y by 8. So in your case the tile position is x=27 and y=5.

If the rotation is anything else than R0, then the rotation as to be applied to the tile position manually (but the division by 8 is still required).

e-music, 2018-11-08:
Rotation is R0, and I did exactly as you said for the x0,y0 coordinates, but nothing is being displayed on the LCD. I did also apply the same calculations for the x1,y1 coordinates as you indicated earlier "(x1+7)/8, (y1+7)/8)", but still nothing happens. Playing/stretching coordinates solves the issue and the symbol is displayed, but I guess I'm refreshing an area much bigger than used by the symbol concerned.

olikraus, 2018-11-08:
> Playing/stretching coordinates solves the issue and the symbol is displayed, but I guess I'm refreshing an area much bigger than used by the symbol concerned.

True, the area is bigger. A tile contains `8*8` pixel.

> By the way, since the symbol is 16*16 pixels, I had to add 16 to each coordinate before division by 8.

yes, also note, that glyphs have there reference point in the lower left corner by default.

e-music, 2018-11-08:
Not sure if I understand what do you mean by "reference point" of glyphs. Anyways, could you please provide an example that shows how to draw a small filled box/rectangle/whatever and have that area only updated using the added functions?

olikraus, 2018-11-08:
The sdl example is linked above. Arduino .ino example is planed. But this implementation is anyways at the beginning.

What exactly does not work? Can you post a more complete example?

olikraus, 2018-11-08:
ok, then there is a missunderstanding. The above mentioned SendWindow command already does the division by 8 for you:
```
void u8g2_SendWindow(u8g2_t *u8g2, u8g2_uint_t x0, u8g2_uint_t y0, u8g2_uint_t x1, u8g2_uint_t y1)
{
  x0 /= 8;
  y0 /= 8;
  x1 += 7;
  x1 /= 8;
  y1 +=7;
  y1 /= 8;
  u8g2_SendTileSubBuffer(u8g2, x0, y0, x1-x0, y1-y0);
}
```
Most of the code just divides the x and y values by 8. My missunderstanding was, that you want to call SendTileSubBuffer directly. One more point, SendWindow requires the upper left and lower right position to describe the window.

So, let us discuss `u8g2_DrawStr(&u8g2, 200, 40, "\xE1");`
The reference point should be the lower left corner of the icon, so 200, 40 is the lower left corner. The glyph size is 16x16, so the upper left corner is 200, 40-16 and the lower right is  200+16,40 (actually i am not sure whether it is 16 or 15... it depense on the exact location of the reference point, but with 16 you are on the safe side...)
So the SendWindow command should be: `u8g2_SendWindow(&u8g2, 200, 40-16, 200+16, 40);`

e-music, 2018-11-10:
Thank you very much. Worked flawlessly!

olikraus, 2018-12-22:
Note: In U8g2 page buffer mode (`_1_` constructor), individual rows can be updated independently.
I have updated the reference manual for this: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbuffercurrtilerow

olikraus, 2019-01-02:
ToDo:
 * copy example to release
 * create documenation

olikraus, 2019-01-30:
current status:
added updateDisplayArea command
u8g2_full_buffer/UpdateArea/UpdateArea.ino
Maybe it should be renamed to sendBufferArea to be more consistent.
Documentation is missing

AndyPeacock, 2019-02-08:
Hi, the updateDisplayArea command is just what I need in my Arduino project. When is it likely to make it into the build available through the Arduino IDE library manager? Is there any way to build my own temporary version that I can use until an official update is released? Sorry for the questions but I'm not sure exactly how all this side of it works. Awesome work on the library by the way. 

Thanks

olikraus, 2019-02-09:
I have created a new beta release. It also includes the above mentioned example. Documentation is still not there...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-02-09:
https://github.com/olikraus/u8g2/wiki/u8g2reference#updateDisplay

closed 2019-02-09 by olikraus

Dabba11, 2019-03-05:
Arduino mega with an ST7920 12864 display full buffer mode. I'm saying nothing more and mentioning this code below (even if it is playing dirty) : 

//clear certain portion of screen (int horizontal_pixel_pos, int vertical_pixel_pos, int width, int height (of portion to be erased))
void setpixels(int a, int b, int c, int d){
          for (int j=0; j<c; j++){
          for(int k=0; k<d; k++){
            u8g2.setDrawColor(0);   //0 to clear pixel i.e black color
            u8g2.drawPixel(a+j,b+k);//in this position
          }
        }
        u8g2.setDrawColor(1); //1 in order to write with white color afterwards again
}