olikraus/u8g2 #737: ESP32 SPI 3 Wire SSD1306 Nois/Snow

arnold-b, 2018-11-03:
Hello, 

I´m trying to use your lib, but all what i get is snow. i tried every type of setup. 
What could i try ?

Thanks Arnold
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.clk   = PIN_CLK;
	u8g2_esp32_hal.mosi  = PIN_MOSI;
	u8g2_esp32_hal.cs    = PIN_CS;
	//u8g2_esp32_hal.dc    = PIN_DC;
	u8g2_esp32_hal.reset = PIN_RESET;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2;  // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_128x64_noname_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_spi_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);   // init u8g2 structure

	u8g2_InitDisplay(&u8g2);

olikraus, 2018-11-03:
The DC pin is required.

arnold-b, 2018-11-03:
Hi, 

 

i also tried this config. 

       u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;

       u8g2_esp32_hal.clk   = PIN_CLK;

       u8g2_esp32_hal.mosi  = PIN_MOSI;

       u8g2_esp32_hal.cs    = PIN_CS;

       u8g2_esp32_hal.dc    = PIN_DC;

       u8g2_esp32_hal.reset = PIN_RESET;

       u8g2_esp32_hal_init(u8g2_esp32_hal);

 

 

same problem

arnold-b, 2018-11-03:
Sent using the free GMX iPad App

olikraus, 2018-11-03:
Maybe there is a connection problem on the DC pin.

olikraus, 2018-11-28:
no further feedback...

closed 2018-11-28 by olikraus