olikraus/u8g2 #738: Add support for Winstar WEO012832 128 x 32 display?

jlaw34, 2018-11-03:
I was wondering if support could be added for the Winstar WEO012832 128 x 32 display? I've posted
init files received from the distributor.

[Winstar WEO012832 init files.zip](https://github.com/olikraus/u8g2/files/2545088/Winstar.WEO012832.init.files.zip)

olikraus, 2018-11-03:
It is impossible to support all the specific displays out there. Instead U8g2 supports specific display chips. In your case, the WEO012832 seems to be driven by an SSD1305 controller.

So my suggestion is to test one of the existing SSD1305 constructors with the same resolution. All available constructors are listed here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

Good candidates are: SSD1305 128X32_NONAME and SSD1305 128X32_ADAFRUIT

olikraus, 2018-12-20:
Closing this, will continue here: # #756

closed 2018-12-20 by olikraus