olikraus/u8g2 #739: Template for a menu

DaveCalaway, 2018-11-04:
Hi everyone,
i would like to create a little menu with this library. I not found an example to do that, so i will try to create something for me and other.
I read the doc, but something goes wrong... can you help me?

- The part in the Setup, Full screen buffer mode, does not show
- The part in the Loop, Page buffer mode (Picture Loop), shows only the Selection List but not the functions.

```
#include <U8g2lib.h>

// All Boards without Reset of the Display
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

bool menu = 0;
uint8_t current_selection = 0;
const char *string_list =
        "NFC\n"
        "RFID\n"
        "Cirrocumulus\n"
        "Cirrostratus";

void setup() {
        u8g2.begin(/*Select=*/ 7, /*Right/Next=*/ A1, /*Left/Prev=*/ A2, /*Up=*/ A0, /*Down=*/ A3, /*Home/Cancel=*/ 8); 

// welcome using Full screen buffer mode
        u8g2.clearBuffer();
        u8g2.setFont(u8g2_font_6x12_tr);
        u8g2.drawStr(15,15,"Tag Initializer");
        u8g2.sendBuffer();
        delay(2000);
}

void loop() {
// Page buffer mode (Picture Loop)
        if(menu==0){
        current_selection = u8g2.userInterfaceSelectionList(
                "Tag Initializer",
                current_selection,
                string_list);
        }

        u8g2.firstPage();
        do {
                if ( current_selection == 0 ) {         // home/cancel button
                        menu = 0;
                }
                if (current_selection == 1) {         // NFC
                       menu = 1;
                       draw_NFC();
                }
                if (current_selection == 2) {         // NFC
                       menu = 1;
                       draw_RFID();
                }
        } while ( u8g2.nextPage() );
        delay(1000);
}

void draw_NFC(){
        u8g2.drawStr(15,15,"AAAA");
        menu = 0;
}

void draw_RFID(){
        u8g2.drawStr(15,15,"BBB");
        menu = 0;
}
```

DaveCalaway, 2018-11-05:
Thx @olikraus 

I did some change in my code, it's much more scalable right now.
The menu do:
- welcome only during the start up ( Full screen buffer mode )
- Tag init menu:
   - NFC
      - draw function ( Page buffer mode (Picture Loop) )
   - RFID
      - RFID menu:
         - Previous menu
         - go to NFC -> draw function ( Page buffer mode (Picture Loop) )
   - Cirrocumulus
   - Cirrostratus

Now the code use:
DATA:    [========  ]  80.0% (used 1638 bytes from 2048 bytes)
PROGRAM: [===       ]  32.8% (used 10590 bytes from 32256 bytes)

How can i reduce the Data and Prog? did i use something function in a wrong way?

```#include <Arduino.h>
#include <U8g2lib.h>

// All Boards without Reset of the Display
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

uint8_t current_selection = 0;

const char *menu1 =
        "NFC\n"
        "RFID\n"
        "Cirrocumulus\n"
        "Cirrostratus";

const char *menu2 =
        "CCC\n"
        "DDD\n"
        "Cirrocumulus\n"
        "Cirrostratus";

//
// This function draw the text with Page buffer mode (Picture Loop)
//
void drawText(byte X,byte Y, const char *text){
        u8g2.firstPage();
        do {
                u8g2.drawStr(X,Y,text);
        } while ( u8g2.nextPage() );
        delay(1000);
}

//
// This function draw a specific menu
//
void select_menu(const char *nameMenu, const char *select_menu){
        current_selection = 0;
        current_selection = u8g2.userInterfaceSelectionList(
                nameMenu,
                current_selection,
                select_menu);
}

//
//
//
void draw_NFC(){
        drawText(15,15,"AAA");
}

//
//
//
void draw_RFID(){
        bool exit = 0;
        while(exit != 1) {
                const char *nameMenu = "RFID Menu";
                select_menu(nameMenu, menu2);

                if ( current_selection == 0 ) {         // home/cancel button
                        exit = 1;
                }
                if (current_selection == 1) {         // NFC
                        draw_NFC();
                }
        }

}

// ====================================================================================
void setup() {
        u8g2.begin(/*Select=*/ 7, /*Right/Next=*/ A1, /*Left/Prev=*/ A2, /*Up=*/ A0, /*Down=*/ A3, /*Home/Cancel=*/ 8);

        // welcome using Full screen buffer mode
        u8g2.clearBuffer();
        u8g2.setFont(u8g2_font_6x12_tr);
        u8g2.drawStr(15,40,"Welcome");
        u8g2.sendBuffer();
        delay(2000);
}

void loop() {
        const char *nameMenu = "Tag Initializer";
        select_menu(nameMenu,menu1);

        if ( current_selection == 0 ) { // home/cancel button
                //
        }
        if (current_selection == 1) {  // NFC
                draw_NFC();
        }
        if (current_selection == 2) {  // RFID
                draw_RFID();
        }
}
```

olikraus, 2018-11-05:
hmm... does not look that much for me.
Of course the `_F_` constructor consumes a lot. The `_1_` requires lesser RAM, but you have to use the FirstPage/NextPage loop instead of sendBuffer() (see the examples)

DaveCalaway, 2018-11-07:
Thx for your help.
I wrote this code, i hope that can be useful for other people. 
Do i need to do a fork and a pull request to add it to the library example?

```
/*
   This is an example of basic menu implemented by U8g2lib.
   For a better usage of ram, we use the Buffer Size mode.

   DaveCalaway
 */

#include <Arduino.h>
#include <U8g2lib.h>

// All Boards without Reset of the Display
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
byte current_selection = 0;

const char *menu1 =
        "NFC\n"
        "RFID\n"
        "Cirrocumulus\n"
        "Cirrostratus";

const char *menu2 =
        "CCC\n"
        "DDD\n"
        "Cirrocumulus\n"
        "Cirrostratus";

//
// This function draw the text
//
void drawText(byte X,byte Y,const char *text){
        u8g2.firstPage();
        do {
                u8g2.drawStr(X,Y,text);
        } while ( u8g2.nextPage() );
        delay(1000);
}

//
// This function draw the menu
//
void select_menu(const char *nameMenu, const char *select_menu){
        current_selection = u8g2.userInterfaceSelectionList(
                nameMenu,
                current_selection,
                select_menu);
        delay(15);         // debounce
}

//
//
//
void draw_NFC(){
        drawText(15,15,"AAAA");
}

//
// RFID menu
//
void draw_RFID(){
        current_selection = 0;
        bool exit = 0;
        while(exit != 1) {
                const char *nameMenu = "RFID Menu";
                select_menu(nameMenu, menu2);

                if ( current_selection == 0 ) {         // home/cancel button
                        exit = 1;
                }
                if (current_selection == 1) {         // NFC
                        draw_NFC();
                }
        }

}

// ====================================================================================
void setup() {
        u8g2.begin(/*Select=*/ 7, /*Right/Next=*/ A1, /*Left/Prev=*/ A2, /*Up=*/ A0, /*Down=*/ A3, /*Home/Cancel=*/ 8);

        // Set font
        u8g2.setFont(u8g2_font_6x12_tr);

        // Welcome
        drawText(15,40,"Welcome");
}

void loop() {
        const char *nameMenu = "Tag Initializer";
        select_menu(nameMenu, menu1);

        if ( current_selection == 0 ) {         // home/cancel button
                //
        }
        if (current_selection == 1) {         // NFC
                draw_NFC();
        }
        if (current_selection == 2) {         // RFID
                draw_RFID();
        }
}
```

closed 2018-12-20 by olikraus