olikraus/u8g2 #740: oled and conversion bmp program

wchpikus, 2018-11-04:
Hi.
One question,what program you using to convert bmp?
I have problem to display graph 256x64 on
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI Oled display.

Nothing on the screen.
I tryed lcdassistant and microlcd java program. Strange is, i only replace jlx display to ssd and forgot how to make this pictures...
I tryed gimp and export, not work.
Problem with 256x64?
When i try old 192x64 picture used on old Progmem, works ok.
Little help);)

olikraus, 2018-11-04:
I always use gimp and export as XBM

wchpikus, 2018-11-04:
Did you rotate the picture or in normal position convert in gimp?

olikraus, 2018-11-05:
Does this matter? But no I didn't rotate the bitmap.

wchpikus, 2018-11-05:
Hmm..strange why do not work for me.
Could you tell me, this manual is ok for export file?

https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/

olikraus, 2018-11-05:
> Could you tell me, this manual is ok for export file?
https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/

Yes, this tutorial should exactly explain all the steps.

wchpikus, 2018-11-05:
Hmm..ok I will today.
So no rotate,convert to b&w..i see;)
I do not use 1 bits pallete..
;)

wchpikus, 2018-11-05:
Ok not work correctly..
[image]
[image]

olikraus, 2018-11-05:
> max resolution is 248x64...fake oled??
250x64 and more do not work...

See here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#16-bit-mode

You may need to enable 16 bit mode for u8g2 (see FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159)

wchpikus, 2018-11-05:
hmm, oki i try:)

wchpikus, 2018-11-05:
ok, it is it:)
great thanks:)
So , only gimp and all works:)

closed 2018-12-20 by olikraus