olikraus/u8g2 #741: Contrast Oled

wchpikus, 2018-11-06:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2
u8g2.setContrast(x);
x from 0 to 255 works, but do not disable oled.
value x=0 still display some data..
Is it possible to dimmer to  "0" this oled?
Regards

olikraus, 2018-11-06:
> Is it possible to dimmer to "0" this oled?

maybe, but at least not with the setContrast command. 

If you want to dimm towards zero brightness, you need to apply several different methods.
There was once a discussion regarding the SSD1306, which could be applied to the SSD1322, but: The register commands are most probably different: #636

wchpikus, 2018-11-06:
i found this:

command(0xC1);//set contrast current
data(0xFF);
command(0xC7);//set master contrast current
data(0x0F);

https://www.lcd-module.de/fileadmin/eng/pdf/grafik/W256064-XALG.pdf

What current is talking about? =>set master contrast current

olikraus, 2018-11-06:
U8g2 will use C1 command. But yes, maybe C7 command will help here.

wchpikus, 2018-11-06:
Is this function will help ( offcource V inside  is data)..
Viod set_current(uint8_t data)
{
  u8x8_cad_StartTransfer(u8g2.getU8x8());
  u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0c7);
  u8x8_cad_SendArg(u8g2.getU8x8(), v << 4);
  u8x8_cad_EndTransfer(u8g2.getU8x8());
}

olikraus, 2018-11-06:
The 0x0c7 command expects data in the lower 4 bits, so it must be:
```
void set_current(uint8_t data)   // data: 0..15
{
u8x8_cad_StartTransfer(u8g2.getU8x8());
u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0c7);
u8x8_cad_SendArg(u8g2.getU8x8(), data & 15);
u8x8_cad_EndTransfer(u8g2.getU8x8());
}
```

> How to read this register after send the new data?

Quote from the SSD1322 datasheet: "Under serial mode, only write operations are allowed." 
The fact that you can not read data from the display controller is very common. As a consequence, U8g2/U8x8 do not include any read command.

wchpikus, 2018-11-06:
Tnx for replay, i will test this isue with dimmer.
Simply animation and resulution will work graet, regulation of brightness will be usefull for me and other users;)
https://youtu.be/HCaQ4OEMig0

olikraus, 2018-11-07:
Nice demo :-)

closed 2018-12-20 by olikraus