olikraus/u8g2 #742: OLED display crashes at random

msigman, 2018-11-06:
I have a Newhaven SSD1322 256x64 OLED that is crashing randomly.  In my code, the display is always being updated in loop() due to horizontal text scrolling, and generally it has no issues. 

However in response to incoming WiFi traffic (when I clear the display and update with a new Artist and Title) it will become corrupted.  Sometimes it comes back to life when a new message arrives, other times it stays dead until I restart.  Sometimes the display flashes 100% pixels full-on and then immediately goes blank.  Sometimes it is random horizontal lines displayed across the screen.  And sometimes it is a corrupted, almost mirror image of what should be displayed, or the colors become inverted (before & after photos below).  Nowhere in my code do I have anything like this.

Any idea what could cause this?  

I've attached the code although this is a fairly large project and will be difficult to parse through -- happy to answer any questions.

**Main ino**
```
#include <ESP8266WiFi.h>
#include <Wire.h>
#include <SPI.h>
#include "Arduino.h"
#include "SonosEsp.h"
#include "Display.h"
#include "Globals.h"

#define MAX_SONOS_PLAYERS 7
#define MAX_WIFI_BUFFER 8000

#define DEBUG false
#define Serial if(DEBUG)Serial
ADC_MODE(ADC_VCC); // Only needed to allow use of ESP.getVCC() in the debug output

unsigned long timeToLevelVolumes = 0;
unsigned long timeToRenewSubscription = 0; // Have to renew subscription every 24 hours
unsigned long lastSubscriptionRenewal = 0; // For debugging
unsigned long timeToTurnOffScreen = 0;
unsigned long timeToStartScrolling = 0; // When a new song beings playing, pause the scrolling for a bit
int _maxContentLength = 16384;
bool scrollingLeft = true;
int titleXCoord = 0;
int newTargetVol = 0; // Global vars tracks last target vol for hysteresis protection
int lastTargetVol = 0;
unsigned long bootupTime = 0;
char ip[21] = "";
char wifiBuffer[MAX_WIFI_BUFFER] = "";

enum state
{
  NOCHANGE = -1,
  STOPPED = 0,
  PLAYING = 1,
  PAUSED = 2,
  TRANSITIONING = 3
};

// Music declarations
state transportState = NOCHANGE;
char title[256] = "";
char artist[64] = "";
int titleLengthInPixels = 0;

SonosEsp sonos;
Display oled;
WiFiServer server(3500);

void setup()
{
  Serial.begin(115200);

  pinMode(2, OUTPUT);    // Blue LED
  digitalWrite(2, HIGH); // Blue LED off

  int progress = 0;

  // Connect to WiFi
  oled.drawSetupMessage("Connecting to WiFi", 0);

[... 488 lines cut ...]
void Display::clear() {
  _display->clearBuffer();
  return;
}

void Display::sleep(bool onoff) {
  _display->setPowerSave(onoff);
  return;
}

void Display::drawSonosLogoLarge() {
  _display->drawXBM(8, 8, 248, 48, sonos_logo_large);
  return;
}

int Display::getTextWidth(const char *text) {
  return _display->getStrWidth(text);
}
```

**Before & After Photos** 
[image]
[image]
[image]
[image]

olikraus, 2018-11-06:
There had been several reports on ESP OS/hardware related issues. Maybe replacing the U8g2 constructor with the software emulated SPI version could be a nice test. It will obviously slow down the graphcis output, but maybe it will help to isolate the problem.

msigman, 2018-11-07:
I tried swapping for SW SPI but it was so much slower than it kept crashing the entire ESP before it could even get through the first few steps, so I don't think I can even carry out that test.

I wonder if I keep it in HW but add some strategic delays of 1-10ms, if that would help.  Are there any areas you suspect may be a culprit that I could target for such a thing?  Or could it be the WiFi radio is consuming enough power that the display signals don't have the voltage?

olikraus, 2018-11-07:
Well, I just wanted to ensure, that this problem is not related to the recent SPI/I2C issues which came up. So all is fine and you can switch back to HW SPI.

Nevertheless, the problem is still not solved. 
At the moment I personally assume a memory issue or any other bugs, not related to u8g2. Many users including myself did use u8g2. I did not encounter any similar bug yet.
U8g2 also does not allocate any dynamic memory, so problems resulting from dynamic memory allocation can not be caused by u8g2.

Maybe your stack size is too less. U8g2 uses a lot of nested calls...

msigman, 2018-11-07:
I agree it's likely not specific to U8g2 itself.  But I do think it's something with the display/SPI communication.  If it was running out of stack space, wouldn't that result in an ESP crash?  In HW mode the ESP functions perfectly normal, but the display itself begins exhibiting bizarre behavior.  Like the SPI commands are not being sent/received properly, in order, or at all.  Can you think of what scenarios would cause ESP to function normally but display to have issues?