olikraus/u8g2 #743: Problem drawing in text mode only (u8x8) SSD1322 256*64 - 4W-SPI Interface...

e-music, 2018-11-07:
Hello everybody! 

@olikraus 
First of all, I would like to thank you for this great contribution. 

Well, my first exposure with u8g2lib was couple of years ago. At the time, I did not invest a lot of time to understand it well. Now that I'm working on a new project, I've been taking this seriously for the last couple months. So far everything works flawlessly with my SSD1322 display in u8g2 full buffered mode. However, because I needed a much faster way to write directly on the display (portion of it) without the need to send a full frame buffer every time I need to update a bunch of pixels, I thought I might consider switching to the text mode "u8x8". 

Anyways, I'm very glad to find out that someone else had exactly the same requirement as mine and a solution is already suggested to overcome the issue. I will follow up regarding the "update the changed pixels" at the corresponding issue here: https://github.com/olikraus/u8g2/issues/736 , but I'd like to know why u8x8 doesn't render anything on the LCD for me. 

As far as I remember, I had this working in the past somehow, but don't remember how. 

This is the setup/init function:
`void LCD_Init_U8x8()
{
	//8x8_Setup(u8x8_d_ssd1322_nhd_256x64, u8x8_cad_011, u8x8_byte_4wire_sw_spi, uC specific);
	u8x8_Setup(&u8x8, u8x8_d_ssd1322_nhd_256x64, u8x8_cad_011, u8x8_byte_4wire_sw_spi, u8x8_gpio_and_delay_stm32 );
	u8x8_InitDisplay(&u8x8);
	u8x8_SetPowerSave(&u8x8, 0);
	u8x8_ClearDisplay(&u8x8);
	u8x8_SetContrast(&u8x8, 60);
}`

And this is drawing function I've been trying:
` u8x8_SetFont(&u8x8, u8x8_font_5x8_f);
  u8x8_DrawString(&u8x8, 60, 30, "Zaher Gharibi");
  u8x8_RefreshDisplay(&u8x8);`

The LCD is initialized, contrast and clear display functions both work without any problem, but nothing is displayed on the LCD. Did I miss anything from the init code or the rendering functions? I guess pages or begin() do not apply here. 

Thank you in advance!

e-music, 2018-11-07:
Thank you very much. You were correct. I changed the coordinates and it works now! 
I was wondering if I can change the rotation in the 8x8 mode?

olikraus, 2018-11-07:
> I was wondering if I can change the rotation in the 8x8 mode? 

you can only use the hardware flip, if supported.

> But how to get values for certain symbols out of the font chosen?

They are listed in the font itself.

[image]

The "ä" umlaut is at hex position e4. In  the string you could write:

`u8x8_DrawString(&u8x8, 1, 1, "\xe4");`

closed 2018-11-10 by e-music