olikraus/u8g2 #744: Cant open font in bdfconv

hoops-doggit, 2018-11-07:
I'm trying to convert the attached bdf font using bdf conv but I just get an error "can not open bdf file". Any idea why?
Below is the bdf
[MyPixel2.zip](https://github.com/olikraus/u8g2/files/2555995/MyPixel2.zip)

olikraus, 2018-11-07:
it works without problems for me

hoops-doggit, 2018-11-07:
Weird... Any chance you could attach the converted file?

olikraus, 2018-11-07:
I used this command:
```
./bdfconv -v -f 1 -m '32-255' MyPixel2.bdf -o mypixel2.c  -n u8g2_font_mypixel2
```
There seem to be some underline problems, which need to be fixed first in the bdf file:

[image]

[mypixel2.c.txt](https://github.com/olikraus/u8g2/files/2556059/mypixel2.c.txt)

hoops-doggit, 2018-11-08:
Thanks for attaching the file. I ran the exact same code using the windows version of bdfconv but for some reason i just get the error: Can not open bdf file 'fontname'

olikraus, 2018-11-08:
> Can not open bdf file 'fontname'.

Hmm I can not say much here. Maybe a lowercase uppercase problem?

> Could you describe the underline problems in the bdf file?

If you look at the picture, for example lowercase "a": The lowest pixel line is missing. Looking into the generated bdf file we see this for the lowercase "a":
```
STARTCHAR 097
ENCODING 97
SWIDTH 504 0
DWIDTH 7 0
BBX 6 5 1 1
BITMAP
78
0C
7C
CC
CC
ENDCHAR
```
It we now convert the bitmap back to pixel, we can see this:
```
78 01111000
0c 00001100
7c 01111100
cc 11001100
cc 11001100
```
So we can clearly see, that the bitmap of the lowercase "a" is incomplete in the bdf file. During export of the bdf file, the original bitmap was truncated.

hoops-doggit, 2018-11-08:
Ahhh thanks for the explaination. That truncation is likely an artifact of Fony's import methods as I found that whenever I imported a ttf into it it would do weird things to my fonts and I'd have to do a lot of clean up. The only characters I'd cleaned up in the file I gave you were some of the numbers.

olikraus, 2018-11-08:
ok....

olikraus, 2018-11-09:
;)